introlab/rtabmap #456: [ERROR] (2019-10-16 10:05:29.952) VWDictionary.cpp:672::addWordRef() Not found word 52780 (dict size=27641)

2017qiuju, 2019-10-17:
[ERROR] (2019-10-16 10:05:29.952) VWDictionary.cpp:672::addWordRef() Not found word 52780 (dict size=27641)

need help ！

matlabbe, 2019-10-21:
Which version are you using? The database may be corrupted, this can be corrected with `rtabmap-recovery` tool

2017qiuju, 2019-10-23:
> Which version are you using? The database may be corrupted, this can be corrected with `rtabmap-recovery` tool

rtabmap version ？the Latest version I used vinsfusion and d435i to generate the binocular map, and then I used rtabmap / bin $. / rtabmap to load it. The above error occurred.

matlabbe, 2019-10-23:
Was it in localization mode? Try the `rtabmap-recovery` tool, and see if when reloading you have the same error.