introlab/rtabmap #457: [ERROR] (2019-10-16 11:38:08.617) OdometryVINS.cpp:489::computeTransform() VINS-Fusion doesn't work with RGB-D data, stereo images are required!

2017qiuju, 2019-10-17:
[ERROR] (2019-10-16 11:38:08.617) OdometryVINS.cpp:489::computeTransform() VINS-Fusion doesn't work with RGB-D data, stereo images are required!

need help

matlabbe, 2019-10-21:
Which camera are you using? You cannot feed RGB-D data to Vins-Fusion, only stereo data.

2017qiuju, 2019-10-23:
> Which camera are you using? You cannot feed RGB-D data to Vins-Fusion, only stereo data.

I use D435i ，when use this  rtabmap/bin$ ./rtabmap

2017qiuju, 2019-10-23:
> > Which camera are you using? You cannot feed RGB-D data to Vins-Fusion, only stereo data.
> 
> I use D435i ，when use this rtabmap/bin$ ./rtabmap

I used vinsfusion and d435i to generate the binocular map, and then I used rtabmap / bin $. / rtabmap to load it. The above error occurred.

matlabbe, 2019-10-23:
With D435i, to use it as a stereo camera, you should select Stereo camera at the top of Source panel in Preferences, then select RealSense2 driver. This will set the realsense in stereo mode.