introlab/rtabmap #459: /home/q/Projects/catkin_ws_vins-fusion/devel/lib/rtabmap_ros/stereo_odometry: error while loading shared libraries: libg2o.so: cannot open shared object file: No such file or directory [ INFO] [1572328227.789534528]: Initializing nodelet with 8 worker threads. process[rtabmap/rtabmap-7]: started with pid [5616] /home/q/Projects/catkin_ws_vins-fusion/devel/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libg2o.so: cannot open shared object file: No such file or directory

2017qiuju, 2019-10-29:
Projects/catkin_ws_vins-fusion/devel/lib/rtabmap_ros/stereo_odometry: error while loading shared libraries: libg2o.so: cannot open shared object file: No such file or directory
[ INFO] [1572328227.789534528]: Initializing nodelet with 8 worker threads.
process[rtabmap/rtabmap-7]: started with pid [5616]
Projects/catkin_ws_vins-fusion/devel/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libg2o.so: cannot open shared object file: No such file or directory

need help ！

matlabbe, 2019-11-03:
Please use a short title for issues. For your error, where libg2o.so is installed?

2017qiuju, 2019-11-12:
thinks i try it !

HaowenLai, 2020-03-15:
I have the same problem. However, for me, the error is 
"./rtabmap: error while loading shared libraries: libgtsam.so.4: cannot open shared object file: No such file or directory".
I build gtsam from source and install it in "/usr/local/lib". I see file libgtsam.so.4 at that path.

2017qiuju, 2020-03-15:
对不起 我没有看到 没能帮到你 深表歉意 这是我的联系方式 qq：2904862402 希望进一步交流

closed 2020-04-02 by matlabbe