introlab/rtabmap #463: RTABmap android build with Realsense 2

TheBricktop, 2019-10-30:
Since there is a way now to connect intel realsense 2 capable cameras with unrooted phones runnning android 7 and up, could it be possible to compile old tango app with realsense v2 support?
If so how to start with building it locally?

matlabbe, 2019-11-03:
If we can use realsense on unrooted phones, yes that could be possible. However, can you refer a link on an example of realsense running on an unrooted phone? The best with a simple code example on how to get the frames. Which cameras are compatible (D435, D435i, T265)?

The modify android app, the longer part is to get a developing environment ready. We can use the docker image: https://github.com/introlab/rtabmap/wiki/Installation#tango-android-development-eclipse

Then we may have to make a realsense2 camera driver for android working with rtabmap. In contrast to Tango, we would need to add the RTAB-Map's odometry thread to compute a pose with D435 cameras.

TheBricktop, 2019-11-04:
I can even provide You with freshly compiled librealsense2 for android, that I've used in Unity project of mine. It kinda works out of the box with realsense sdk in unity. Currently D435 and D435i are compatible, but t265 could be compatible if i manage to compile the hacky driver for it on android. ( its [here](https://github.com/InstytutXR/SlamDriver) if You want it ) for example there is unofficial realsense app on [gplay ](https://play.google.com/store/apps/details?id=com.intel.realsense.camera&hl=pl)that runs out of the box with d435.
As for odometry I am very much looking into porting RTABMAP vo for and relocalisation to android (or even unity, im like a broken record) cause ARcore has stagnated so much that we are resorting to adding external vslam cameras to achieve something similar.

TheBricktop, 2020-02-15:
Im currently working on a simple app combining the realsense depth camera solution and the ARcore capabilities to create a simple point cloud  in Unity will post the results

matlabbe, 2020-05-25:
Note that RTAB-Map tango app has been recently refactored to be independent of Google Tango. It works with ARCore and AREngine. There are new Camera drivers here: https://github.com/introlab/rtabmap/tree/master/app/android/jni.

TheBricktop, 2020-05-26:
so it could work in tandem with realsense camera? 
it could be awesome!

cerrajeriasabaris, 2024-06-22:
Hello, did you manage to get realsense working

parkerlreed, 2024-08-11:
Would be great to have the realsense drivers available for the Android port.