introlab/rtabmap #464: rtabmap make error

vnsiot, 2019-11-04:
Hi,
While building RTABMAP i'm facing the errors, i have attached the screenshot below please help me with this
[image]

Thanks in advance.

matlabbe, 2019-11-04:
Please copy/paste the text instead of taking a screenshot, it will be easier for search engines to find the issue afterwards:
```
undefined reference to `pcl::search::Search<pcl::PointNormal>::getName() const'
undefined reference to `pcl::search::Search<pcl::PointXYZRGBNormal>::getName() const'
undefined reference to `pcl::search::Search<pcl::PointXYZRGB>::getName() const'
undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName() const'
```
 
Some related questions:

- http://official-rtab-map-forum.67519.x6.nabble.com/Compiling-in-ROS-Indigo-with-PCL-1-7-2-Undefined-reference-to-td386.html
- https://github.com/introlab/rtabmap/issues/127
- https://github.com/introlab/rtabmap_ros/issues/18
- https://github.com/introlab/rtabmap/issues/104
- https://github.com/introlab/rtabmap/issues/450
- https://github.com/introlab/rtabmap_ros/issues/322

How PCL is installed? What is your system?

vnsiot, 2019-11-11:
Thanks for the quick reply.
I have followed the above links and  error got resolved after building the pcl-1.7.2 from source.

closed 2019-11-13 by vnsiot