introlab/rtabmap #465: How can I see the trajectory?

1730189, 2019-11-06:
I was wondering if there's a way to see and work with the trajectory outside of rtabmap that was created while making a map using a Kinect v1with OpenNI-PCL. 
Thanks in advance

matlabbe, 2019-11-06:
File -> Export poses...

1730189, 2019-11-06:
Sorry for the beginners question.
Is there a document for me to read where I can know what does the data mean?
Thanks a lot for your help.

matlabbe, 2019-11-06:
There is no official documentation about all options in the interface, you may look at this tutorial covering the basic parts:
https://github.com/introlab/rtabmap/wiki/Kinect-mapping

For your question, there is a similar one here:
http://official-rtab-map-forum.67519.x6.nabble.com/Export-pose-for-every-frame-td750.html