introlab/rtabmap #467: questions about download Bovisa dataset

LiYingOne, 2019-11-09:
BovisaOutdoor-2008-10-04 dataset size is 2277,but this link http://www.rawseeds.org/rs/capture_sessions/view/13 , I downloaded Bovisa_2008-10-04-FRONTAL.tar.torrent . it has 68288 images.How can I get right BovisaOutdoor-2008-10-04 dataset. And How can I get right Bovisa Mixed-2008-10-06 dataset.

matlabbe, 2019-11-17:
I think only 1 image of 30 was kept for the loop closure experiment. The Bovisa dataset has image at 30 Hz, but the experiment was done at 1 Hz. If you do `68288/2277 = ~30`.

dzqiu, 2020-04-24:
Hello, I can't download Bovisa dataset from the torrent. Anyone can help me? Thanks.

Sologala, 2020-12-31:
HI, I can't get thie dataset too due to that this torrent seem like out of source.  Please contact me if you have idea for it. 
dawsonwen123@gmail.com 

Thanks, Best wish.

xzwang12345, 2021-04-01:
Hi @LiYingOne I can't get the dataset too, If you have this dataset,can you share it? Thanks

janm31415, 2022-05-04:
You can download the dataset via https://airlab.deib.polimi.it/datasets-and-tools/