introlab/rtabmap #469: rtabmap crashes with null pointer in rs2::error

salvato, 2019-11-12:
Hi,
I've just installed rtabmap (without ros) on a fresh installation of Ubuntu 18.04.3 LTS (kernel 5.0.0-32-generic) with a RealSense D415 connected to a USB 3.0 port. 

Starting a new acquisition with all default settings results in a crash with the following reason:
terminate called after throwing an instance of 'rs2::error'
  what():  null pointer passed for argument "from".

If I disable the IR emitter and use "IR mode" for tracking in the "Preferences" then the program run but no rgb images are captured.

matlabbe, 2019-11-17:
Can you set log level to debug (Preferences->Logger) and copy paste the log just before it crashes? What is the realsense driver version?

salvato, 2019-11-19:
Hi. The last lines on the LogRtabmap.txt file are:

> [DEBUG] (2019-11-19 14:01:30.133) MainWindow.cpp:4881::startDetection() 
> [DEBUG] (2019-11-19 14:01:30.134) MainWindow.cpp:4951::startDetection() 
> [DEBUG] (2019-11-19 14:01:30.134) PreferencesDialog.cpp:5296::createCamera() 
> [DEBUG] (2019-11-19 14:01:30.135) CameraRealSense2.cpp:82::CameraRealSense2() 
> [DEBUG] (2019-11-19 1
> 

The file is clearly interrupted before flushing the last content.
The About Dialog of RTAB-Map report also:

```
Version 0.19.6

OpenCV version 3.2.0
PCL version 1.9.1
VTK version 7.1.1
Qt verion 5.12.4
.
.
.
RealSense2: YES
.
.
.
```
The installed realsense libraries are:

```
librealsense2-dev/xenial,now 2.30.0-0~realsense0.1693 amd64 [installato]
librealsense2-dkms/xenial,now 1.3.7-0ubuntu1 all [installato]
librealsense2-gl/xenial,now 2.30.0-0~realsense0.1693 amd64 [installato, automatico]
librealsense2-udev-rules/xenial,now 2.30.0-0~realsense0.1693 amd64 [installato, automatico]
librealsense2-utils/xenial,now 2.30.0-0~realsense0.1693 amd64 [installato]
librealsense2/xenial,now 2.30.0-0~realsense0.1693 amd64 [installato, automatico]
```

matlabbe, 2019-11-20:
Unfortunately I cannot reproduce the problem with my D435i. I don't have a D415 for testing, though it seems to be okay from this last year post: https://github.com/introlab/rtabmap/issues/308 (same function I use for D435 and D435i should work)

I tested with:
```
librealsense2-dev:
  Installed: 2.30.0-0~realsense0.1694
```

It seems crashing after the constructor, it should be in the [CameraRealSense2::init()](https://github.com/introlab/rtabmap/blob/b39fd0321ee8a0c683019dfc02141506615a0df3/corelib/src/camera/CameraRealSense2.cpp#L401) function. I would expect to see more logs before crashing though before realsense functions are called. There is maybe one of the realsense function that is asserting an error that is not caught. Try logging in terminal instead of a file, you may get more precisely the last log before it crashes.

ntreepoint, 2020-05-19:
Hi,
I have exactly the same issue when using the Realsense D415. I use the master and i build and run on Ubuntu 20.04.
Is there any update on this issue so that i can fix it?

matlabbe, 2020-05-19:
Show log as https://github.com/introlab/rtabmap/issues/469#issuecomment-554700540

ntreepoint, 2020-05-20:
I have the same kind of logs, write is interrupted before getting any trace from camera part:

> [DEBUG] (2020-05-20 09:49:50.667) PreferencesDialog.cpp:4070::setParameter() g2o/PixelVariance=1
[DEBUG] (2020-05-20 09:49:50.667) PreferencesDialog.cpp:4070::setParameter() g2o/RobustKernelDelta=8
[DEBUG] (2020-05-20 09:49:50.667) PreferencesDialog.cpp:4070::setParameter() g2o/Solver=0
[DEBUG] (2020-05-20 09:49:58.265) PreferencesDialog.cpp:2540::writeSettings() _obsoletePanels=8 modified parameters=0
[DEBUG] (2020-05-20 09:49:58.265) MainWindow.cpp:4206::applyPrefSettings() 
[DEBUG] (2020-05-20 09:49:59.516) PreferencesDialog.cpp:1610::closeDialog() 
[DEBUG] (2020-05-20 09:50:08.606) MainWindow.cpp:5025::startDetection() 
[DEBUG] (2020-05-20 09:50:08.606) MainWindow.cpp:5097::startDetection() 
[DEBUG] (2020-05-20 09:50:08.606) PreferencesD 

> 

Please note that i have tried on two different configurations: one is ubuntu 20.04 with default realsense drivers. The other one is ubuntu 18.04 with realsense drivers recompiled from up-to-date SDK. With both configurations realsense viewer is working fine whereas rtabmap crashes.

matlabbe, 2020-05-20:
I am surprised we don't even get this log from the constructor: https://github.com/introlab/rtabmap/blob/89ece8cd4eb994b02852206d2fda99e298548c24/corelib/src/camera/CameraRealSense2.cpp#L83

If it is crashing in the constructor, can you show what is rtabmap dependencies? One trick is to disable in cmake most of the dependencies to see which one has conflicts with realsense. 

You can also run in gdb to have more details on where it crashes:
```
$ gdb rtabmap
```
Type "run" to start rtabmap.
When it crashes, type "bt" to show the backtrace.

ntreepoint, 2020-05-22:
Hi,
I have investigated using GDB. The crash is triggered by line 764:
`*depthToRGBExtrinsics_ = depthStreamProfile.get_extrinsics_to(rgbStreamProfile);`
Exact error is "Cannot create a lazy string with address 0x0, and a non-zero length.><error reading variable: Cannot create a lazy string with address 0x0, and a non-zero length"
This line was only usefull for the dual mode (D400+ T265), which is a mode i don't use. So i have remooved this line. Then the initialization seems successfull. But i don't get any frame from the camera, rtamap states "The camera has reached end of stream", which is weirs as i can see frame callbacks in the logs.
Here are the logs

[LogRtabmap.zip](https://github.com/introlab/rtabmap/files/4668443/LogRtabmap.zip)

ntreepoint, 2020-05-22:
Hi,
I have found the issue. When setting the correct profile for each sensor, the profile set for the stereo module was not correct. An infrared profile was set instead of a depth one. I have patched line 646 to add a stream type condition like this:
`if((!ir_ && video_profile.format() == RS2_FORMAT_RGB8 && video_profile.stream_type() == RS2_STREAM_COLOR) ||`

matlabbe, 2020-05-25:
I don't have access to my D435 or D435i camera right now. If someone can test if this change doesn't break D435/D435i, I will push the change.

kneehit, 2020-07-22:
I am encountering the same issue as well. Ubuntu 18.04 and Realsense D415