introlab/rtabmap #472: make errors occured after "cmake .. -DQVTK_INCLUDE_DIR=/usr/include -DQVTK_LIBRARY=/usr/lib " and "make"

zengletian1491, 2019-11-14:
make errors occured after "cmake .. -DQVTK_INCLUDE_DIR=/usr/include -DQVTK_LIBRARY=/usr/lib " and "make", the detailed information is presented as follows:
**Work/rtabmap/corelib/src/util3d_surface.cpp:61:67: fatal error: pcl/surface/vtk_smoothing/vtk_mesh_quadric_decimation.h: No such file or directory
compilation terminated.
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:639: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/util3d_surface.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/util3d_surface.cpp.o] Error 1
CMakeFiles/Makefile2:266: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2**
Could you please give me some suggestions?Thanks very much!

zengletian1491, 2019-11-15:
In fact, vtk_mesh_quadric_decimation.h is in the file with path **"/usr/include/pcl-1.7/pcl/surface/vtk_smoothing/vtk_mesh_quadric_decimation.h".**

matlabbe, 2019-11-17:
Strange, `/usr/include/pcl-1.7` should be in the include paths when building rtabmap. Can you show output of `make VERBOSE=1`?

zengletian1491, 2019-11-19:
The results is shown in the following link:
https://github.com/introlab/rtabmap/issues/474
Thanks very much!