introlab/rtabmap #473: OdometryVINS.cpp:353::computeTransform() Waiting an image for initialization...

2017qiuju, 2019-11-14:
i use vinsfusion and realsenseD435i make Binocular Stereo .db file ,
in the commod i run rtabmap/bin$ ./rtabmap
to load the db file , in the  Preferences i set stereo and imufilter but i relocation is false
put tihs error please hellp me !

have VINS-Fusion-Stereo.db
/rtabmap/bin$ ./rtabmap
[ WARN] (2019-11-14 17:42:44.963) OdometryVINS.cpp:353::computeTransform() Waiting an image for initialization...

matlabbe, 2019-11-17:
So you are trying to start a new session from a previous database, the warning tells that no images have been received yet. CAn you set logger in debug mode to have more info? (Preferences->Logger)

2017qiuju, 2019-11-18:
Hello, I want to use vins-fusion to do Stereo relocation. The camera I use is realsenseD435i. My data file is collected with test_d435ivio.launch startup file, then run /rtabmap/bin$ ./rtabmap
But always prompted this warning:
[ WARN] (2019-11-14 17:42:44.963) OdometryVINS.cpp:353::computeTransform() Waiting an image for initialization...
The output of the logger I have set is Debug and then outputs the following information:
[ WARN] (2019-11-18 11:28:12.823) OdometryVINS.cpp:353::computeTransform() Waiting an image for initialization...
[DEBUG] (2019-11-18 11:28:12.823) OdometryThread.cpp:120::mainLoop() Odom pose = xyz=[null] rpy=[null]
[DEBUG] (2019-11-18 11:28:12.823) RtabmapThread.cpp:361::handleEvent() OdometryEvent
[DEBUG] (2019-11-18 11:28:12.824) MainWindow.cpp:967::processOdometry()
[DEBUG] (2019-11-18 11:28:12.824) MainWindow.cpp:991::processOdometry() odom lost
[DEBUG] (2019-11-18 11:28:12.824) ImageView.cpp:899::setFeatures() depth=640x480 sceneRect=640.000000x480.000000
[DEBUG] (2019-11-18 11:28:12.824) ImageView.cpp:902::setFeatures() xRatio=1.000000 yRatio=1.000000
[DEBUG] (2019-11-18 11:28:12.827) CameraRealSense2.cpp:210::imu_callback() GYRO callback! 1574047692827.407959 (-0.000006 -0.001739 0.005254)
[DEBUG] (2019-11-18 11:28:12.832) CameraRealSense2.cpp:210::imu_callback() GYRO callback! 1574047692832.409912 (-0.003496 0.000006 0.000018)
[DEBUG] (2019-11-18 11:28:12.835) CameraRealSense2.cpp:261::frame_callback() Frame callback! 1574047692835.780762
[DEBUG] (2019-11-18 11:28:12.837) CameraRealSense2.cpp:210::imu_callback() GYRO callback! 1574047692837.666504 (-0.000006 0.008733 0.001764)
[DEBUG] (2019-11-18 11:28:12.838) CameraRealSense2.cpp:210::imu_callback() ACC callback! 1574047692838.371582 (-0.200062 -9.836268 0.622711)
[DEBUG] (2019-11-18 11:28:12.842) CameraRealSense2.cpp:210::imu_callback() GYRO callback! 1574047692842.657227 (-0.000006 0.001752 0.005254)
[DEBUG] (2019-11-18 11:28:12.843) CameraRealSense2.cpp:261::frame_callback() Frame callback! 1574047692842.898682
[DEBUG] (2019-11-18 11:28:12.843) CameraRealSense2.cpp:909::captureImage() Frameset arrived.
[DEBUG] (2019-11-18 11:28:12.847) CameraRealSense2.cpp:210::imu_callback() GYRO callback! 1574047692847.629883 (-0.000006 -0.005230 0.000018)
[DEBUG] (2019-11-18 11:28:12.852) CameraRealSense2.cpp:210::imu_callback() GYRO callback! 1574047692852.655518 (-0.001751 -0.008720 0.003509)
[DEBUG] (2019-11-18 11:28:12.854) CameraRealSense2.cpp:210::imu_callback() ACC callback! 1574047692854.396973 (-0.240139 -9.754127 0.631015)
[DEBUG] (2019-11-18 11:28:12.855) Camera.cpp:106::takeImage() Time capturing image = 0.032192s
[DEBUG] (2019-11-18 11:28:12.855) CameraThread.cpp:434::postUpdate() -0.544851 0.416367 -0.429281 0.587786 (gyro=-0.000006 0.001413 0.005000, acc=-0.211384 -9.813064 0.625057, 1574047692.843048s)
[DEBUG] (2019-11-18 11:28:12.855) CameraThread.cpp:142::mainLoop()
[DEBUG] (2019-11-18 11:28:12.855) OdometryThread.cpp:116::mainLoop() Processing data...
Can I use vins-fusion to do Stereo's relocation? Is there a related document? I can refer to it.

matlabbe, 2019-11-18:
If you start from a fresh database, does it work? Here is my Source parameters with RGB-D realsense2 mode (and using all default parameters unless VINS as Odometry approach and its config file):
[image]
Note the "Publish inter IMU..." checkbox, which is required for VINS.

You should be able to relocalize independently of the odometry approach. IT is however recommended to use a similar camera than the one using to create the map. Did you record a stereo database with the launch file or a RGB-D one?

2017qiuju, 2019-11-19:
Can you tell me your email address? I'd like to send you the video of my operation. I think it can help solve my problem，my  email address is 2904862402@qq.com

2017qiuju, 2019-11-29:
If you have time to get video from my 2904862402@qq.com to help me?

matlabbe, 2019-12-01:
Sorry for the delay, I got quite busy last week. VINS won't work with RGB-D data. Update rtabmap to get access to  D435 in stereo IR mode option and to enable inter imu publishing:

[image]

2017qiuju, 2019-12-02:
I do it but not work ! thank for you hellp!