introlab/rtabmap #474: when “make” was typed in, many strange errors, that had never existed,  occurred as follows

zengletian1491, 2019-11-18:
when “make” was typed in, many strange errors, that had never existed,  occurred as follows:

/usr/local/include/pcl-1.9/pcl/point_types.h: In function ‘const pcl::_PointDEM& **pcl::common::operator+=(pcl::_PointDEM&, const float&)’:
/usr/local/include/pcl-1.9/pcl/point_types.h:698:1: error: ‘plusscalar’ is not a member of ‘pcl::traits’
 POINT_CLOUD_REGISTER_POINT_STRUCT (pcl::_PointDEM,
 ^
/usr/local/include/pcl-1.9/pcl/point_types.h:698:1: error: ‘plusscalar’ is not a member of ‘pcl::traits’
 POINT_CLOUD_REGISTER_POINT_STRUCT (pcl::_PointDEM,
 ^**

**/usr/local/include/pcl-1.9/pcl/point_types.h:874:33: error: ‘enable_if_t’ in namespace ‘std’ does not name a template type
     using HasNoIntensity = std::enable_if_t<!has_intensity_v<PointT>, bool>;
                                 ^
/usr/local/include/pcl-1.9/pcl/point_types.h:886:27: error: ‘enable_if_t’ in namespace ‘std’ does not name a template type
     using HasColor = std::enable_if_t<has_color_v<PointT>, bool>;
                           ^**

**/usr/local/include/pcl-1.9/pcl/common/point_tests.h:120:38: error: ‘pcl::traits::HasNoNormal’ has not been declared
   template <typename PointT, traits::HasNoNormal<PointT> = true> constexpr inli
                                      ^
/usr/local/include/pcl-1.9/pcl/common/point_tests.h:120:49: error: expected ‘>’ before ‘<’ token
   template <typename PointT, traits::HasNoNormal<PointT> = true> constexpr inli**

**/usr/local/include/pcl-1.9/pcl/common/impl/copy_point.hpp:90:28: error: ‘enable_if_t’ is not a member of ‘std’
                            std::enable_if_t<boost::mpl::and_<boost::mpl::not_<s**

**/usr/local/include/pcl-1.9/pcl/PolygonMesh.h:48:48: error: request for member ‘vertices’ in ‘polygon’, which is of non-class type ‘int’
                                        polygon.vertices.end (),
                                                ^**

**/usr/local/include/pcl-1.9/pcl/io/file_io.h:317:8: error: ‘enable_if_t’ in namespace ‘std’ does not name a template type
   std::enable_if_t<std::is_integral<Type>::value, bool>
        ^**

**/usr/local/include/pcl-1.9/pcl/common/io.h:65:77:   required from here
/usr/include/c++/5/bits/predefined_ops.h:234:30: error: no match for call to ‘(pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>) (const pcl::PCLPointField&)’
  { return bool(_M_pred(*__it)); }
                              ^**

Please give me some suggestions! Thank you!

matlabbe, 2019-11-19:
It seems PCL 1.9 is using C++14: https://en.cppreference.com/w/cpp/types/enable_if

We may have to adjust the cmake build directives for PCL 1.9, similar to this:
https://github.com/introlab/rtabmap/blob/5f12459437d07aef772ead2ce48a0bd8412ccf07/CMakeLists.txt#L568
Try adding this line at the global scope [here](https://github.com/introlab/rtabmap/blob/5f12459437d07aef772ead2ce48a0bd8412ccf07/CMakeLists.txt#L561) in the main `CMakeLists.txt` to build with c++14 support.

zengletian1491, 2019-11-19:
Dear matlabbe, I tried adding "set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14") " as you said, but it did not work!

zengletian1491, 2019-11-19:
Using pcl source code with the link **"https://github.com/PointCloudLibrary/pcl/releases"**, choosing **source code ("pcl-pcl-1.9.1.tar.gz") installation**, these errors could be avoided, but other problems still existed in the following link **"https://github.com/introlab/rtabmap/issues/475"**.

zengletian1491, 2019-11-19:
Very Strange!