introlab/rtabmap #475: After "make" was typed in, errors occurred as follows:

zengletian1491, 2019-11-19:
After "make" was typed in, errors occurred as follows:

[ 20%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_surface.cpp.o
In file included from /usr/local/include/boost/detail/endian.hpp:9:0,
                 from /usr/local/include/pcl-1.9/pcl/PCLPointCloud2.h:11,
                 from /usr/local/include/pcl-1.9/pcl/PolygonMesh.h:10,
                 from /home/zeng/Work/rtabmap/corelib/src/../include/rtabmap/core/util3d_surface.h:33,
                 from /home/zeng/Work/rtabmap/corelib/src/util3d_surface.cpp:28:
/usr/local/include/boost/predef/detail/endian_compat.h:11:161: note: #pragma message: The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead
 #pragma message("The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead")
                                                                                                                                                                 ^
In file included from /usr/local/include/boost/config/detail/suffix.hpp:543:0,
                 from /usr/local/include/boost/config.hpp:61,
                 from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:17,
                 from /usr/local/include/boost/shared_ptr.hpp:17,
                 from /usr/local/include/pcl-1.9/pcl/PCLHeader.h:10,
                 from /usr/local/include/pcl-1.9/pcl/PolygonMesh.h:9,
                 from /home/zeng/Work/rtabmap/corelib/src/../include/rtabmap/core/util3d_surface.h:33,
                 from /home/zeng/Work/rtabmap/corelib/src/util3d_surface.cpp:28:
**/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_half<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:273:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(half, 5.000000000000000000000000000000000000e-01, "5.00000000000000000000000000000000000000000000000000000000000000000000000000000000000000
   ^
/usr/local/include/boost/math/constants/constants.hpp:273:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_third<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:274:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(third, 3.333333333333333333333333333333333333e-01, "3.3333333333333333333333333333333333333333333333333333333333333333333333333333333333333
   ^
/usr/local/include/boost/math/constants/constants.hpp:274:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_twothirds<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:275:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(twothirds, 6.666666666666666666666666666666666666e-01, "6.666666666666666666666666666666666666666666666666666666666666666666666666666666666
   ^
/usr/local/include/boost/math/constants/constants.hpp:275:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_two_thirds<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:276:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(two_thirds, 6.666666666666666666666666666666666666e-01, "6.66666666666666666666666666666666666666666666666666666666666666666666666666666666
   ^
/usr/local/include/boost/math/constants/constants.hpp:276:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_sixth<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:277:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(sixth, 1.666666666666666666666666666666666666e-01, "1.6666666666666666666666666666666666666666666666666666666666666666666666666666666666666
   ^
/usr/local/include/boost/math/constants/constants.hpp:277:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_three_quarters<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:278:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(three_quarters, 7.500000000000000000000000000000000000e-01, "7.5000000000000000000000000000000000000000000000000000000000000000000000000000
   ^
/usr/local/include/boost/math/constants/constants.hpp:278:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_root_two<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:279:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(root_two, 1.414213562373095048801688724209698078e+00, "1.4142135623730950488016887242096980785696718753769480731766797379907324784621070388
   ^
/usr/local/include/boost/math/constants/constants.hpp:279:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_root_three<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:280:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(root_three, 1.732050807568877293527446341505872366e+00, "1.73205080756887729352744634150587236694280525381038062805580697945193301690880003
   ^
/usr/local/include/boost/math/constants/constants.hpp:280:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_half_root_two<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:281:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(half_root_two, 7.071067811865475244008443621048490392e-01, "7.07106781186547524400844362104849039284835937688474036588339868995366239231053
   ^
/usr/local/include/boost/math/constants/constants.hpp:281:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_ln_two<T>::get(const mpl_::int_<5>&)’:
/usr/local/include/boost/math/constants/constants.hpp:282:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(ln_two, 6.931471805599453094172321214581765680e-01, "6.931471805599453094172321214581765680755001343602552541206800094933936219696947156058
   ^
/usr/local/include/boost/math/constants/constants.hpp:282:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/local/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_ln_ln_two<T>::get(const mpl_::int_<5>&)’:
[... 330 lines cut ...]
/usr/include/c++/5/bits/stl_map.h:688:9: note:   template argument deduction/substitution failed:
/home/zeng/Work/rtabmap/corelib/src/util3d_surface.cpp:2404:25: note:   candidate expects 2 arguments, 1 provided
       imageSize.height)));
                         ^
[... cut ...]
       setPolynomialFit (bool polynomial_fit)
       ^
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:639: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/util3d_surface.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/util3d_surface.cpp.o] Error 1
CMakeFiles/Makefile2:266: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2**

matlabbe, 2019-11-20:
You also built boost from source... It will be hard to debug this here without having your custom configuration. If you can create a Dockerfile with a all steps used to install the current setup you have, we may take a look. What I can test is trying PCL 1.9 from source at least, but all other dependencies I would recommend to use the binaries. I'll give a try with PCL 1.9 in a docker image and come back to you.

matlabbe, 2019-11-21:
I tried building PCL 1.9 on this docker image: introlab3it/rtabmap:18.04
```
$ docker run -it --rm introlab3it/rtabmap:18.04
# apt remove libpcl-dev

# git clone https://github.com/PointCloudLibrary/pcl.git
# cd pcl 
# mkdir build
# cd build
# cmake ..
# make -j8
# make install

# cd 
# git clone https://github.com/introlab/rtabmap.git
# mkdir build
# cd build
# cmake ..
# make -j8
```

Only issue I had is fixed now by this [commit](https://github.com/introlab/rtabmap/commit/85e7c9cf0481d2802af2708f184dfa34944930b6). How this docker image is built can be found in this Dockerfile: https://github.com/introlab/rtabmap/blob/master/docker/xenial/Dockerfile

cheers,
Mathieu

zengletian1491, 2019-11-22:
I compiled "rtabmap" as you directed and it succeeded. Thanks very much for your help!

closed 2019-11-22 by matlabbe