introlab/rtabmap #476: Kinect for Azure option not enabled

scimad, 2019-11-19:
Why is Kinect for Azure option not active in my RTAB-Map? I learned that RTAB-Map only supports video captured from Azure Kinect for now, but I have that option disabled in Detection >> Select Source >> More Options >> RGBD ?

System Configuration: 
Ubuntu 18.04

Further info: I also confirm the applications / examples on Azure/SDK running, and RTAB map also running with Intel D435i.

matlabbe, 2019-11-19:
When building rtabmap, make sure you see this at the end of the cmake output:
```bash
$ cd rtabmap/build
$ cmake ..
...
--  Camera Drivers:
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = YES (With CUDA)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = YES (License: Apache-2)
...
```
it should be `With Kinect for Azure     = YES` if cmake is able to find K4A. Are you on Windows? We didn't do a lot of testing with K4A (we cannot buy it from Canada yet), maybe this cmake [script](https://github.com/introlab/rtabmap/blob/master/cmake_modules/FindK4A.cmake) would have to be updated on your system if K4A is installed and cmake shows `NO`.

cheers,
Mathieu

scimad, 2019-11-20:
Works perfectly, thanks.

closed 2019-11-20 by scimad