introlab/rtabmap #478: After " cmake -D_WITH_ALICE_VISION=OFF .." and "make" were typed in Ubuntu 16.04 LTS, Errors occurred in the "make" stage as follows!

zengletian1491, 2019-12-02:
After " cmake -DWITH_ALICE_VISION=OFF .." and "make" were typed in Ubuntu 16.04 LTS, Errors occurred in the "make" stage as follows(the two following command lines about vtk are also typed in: 
sudo apt-get install libvtk5-dev
sudo apt-get install libvtk5-qt4-dev ):

**[ 42%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/imufilter/MadgwickFilter.cpp.o
[ 42%] Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: cannot find -lvtkGUISupportQt
collect2: error: ld returned 1 exit status
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1964: recipe for target '../bin/librtabmap_core.so.0.19.6' failed
make[2]: *** [../bin/librtabmap_core.so.0.19.6] Error 1
CMakeFiles/Makefile2:266: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2**

I'm not sure how to solve this problem.Could you please give me some suggestions? The detailed 
results of "cmake -DWITH_ALICE_VISION=OFF .." and "make" are shown in the attached files.

[cmake-make.tar.gz](https://github.com/introlab/rtabmap/files/3911203/cmake-make.tar.gz)

matlabbe, 2019-12-02:
vtkGUISupportQt may have been added by this line:
https://github.com/introlab/rtabmap/blob/9582ea4b555e2cdc1f24ee7941c882e38156d31c/CMakeLists.txt#L293

It is difficult to know which vtk/Qt versions are installed on your computer and those used by default. Normally on Xenial, it is Qt5 (qtbase5-dev) and vtk6 (libvtk6-qt-dev) I think by default.

zengletian1491, 2019-12-26:
solution:
in /usr/local/lib/cmake/GTSAM/GTSAM-exports-release.cmake, Line 41：
change "Boost::timer"  to
"/usr/lib/x86_64-linux-gnu/libboost_timer.so".

zengletian1491, 2019-12-26:
Using pcl-1.7 may solve this problem! 
On Ubuntu16.04 LTS, pcl-1.7 can be installed with the following commands:
sudo apt-get install libpcl-dev.