introlab/rtabmap #479: Loop closure detection questions

chiaueng, 2019-12-04:
Dear Sirs,
As a beginner, I was trying to debug the poor loop closure detection issue. I am using ZED camera with the DetectorStrategy of ORB. I tried to go around the same place twice(The purpose is to see the loop closure detection performances). The database is at : https://drive.google.com/open?id=13bEJelG8eI_BhjcrwC4f0O2QiZLfYj6E

The tool I used to browse the data is rtabmap-databaseViewer. What I observed are : 
(1) Sometimes there are no features detected. For example, index 983 to 944.
(2) Sometimes the loop closure can't be detected. For example, index 979 to 984 are failed to detect the loop closure while index 973~978 can detect the loop closure successfully.

May I have your suggestions which directions to check these problems ? And how can I get better results ? Thanks in advance.

Sincerely,
Chiaueng

matlabbe, 2019-12-13:
Hi Chiaueng,

Unfortunately, the range of the ZED camera is short. You could use the stereo version of the ZED  (left/right) instead of the RGB-D version (rgb/depth) to extract features farther, rtabmap will work better in stereo mode. Which odometry are you using? The trajectory seems to drift a lot.

For loop closures not detected, it is mostly because only features on road are used. Using stereo mode will extract features not just on the road, more discriminative features are better. If you are going to use ORB detector, use the new version ORB-OCTREE (more similar to what is used in ORB_SLAM2), use `Kp/DetectorStrategy=10` and `Vis/FeatureType=10`. ORB features extracted will be more uniformly distributed in the image, thus increasing accuracy when computing transformations.

ORB (Kp/DetectorStrategy=2):
[image]
ORB-OCTREE (Kp/DetectorStrategy=10):
[image]

Note that if the images were **stereo**, there would features everywhere in the image.

cheers,
Mathieu

chiaueng, 2019-12-16:
Hi Mathieu,
Thanks for your suggestion. After changing to use stereo version by following this wiki (http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping),  I can see more features are extracted and get a better loop closure result. I also changed to use Kp/DetectorStrategy=10 and it looks the ORB features are extracted more uniformly distributed in the image.

For the trajectory drift issue, I am using the odometry topic of /zed/zed_node/odom. Could you suggest me how to improve it ? I found using stereo version can improve it, but I am not clear its relationship.

By the way, it seems using stereo version will cause "Lost Odometry" occasionally. I didn't see this happen in RGB-D version at the same environment. Is there a way to avoid this problem ? The suggestion on wiki is to replace the camera to the last valid position. However, for my usage case, it is hard to replace the camera to the last valid position. For example, if I use it for Camera SLAM on Auto Car, it's impossible to go back to the previous place while driving on the road.

Sincerely,
Choaueng

matlabbe, 2019-12-19:
Hi, if you have "Lost Odometry" occasionally and you are using `/zed/zed_node/odom` as input odometry, it is not related to stereo mode on rtabmap side. You may had the same problem before. Zed odometry is somewhat resetting its odometry. If it recovers automatically, it could be a minor problem though.

You could compare zed odometry with [rtabmap_ros/stereo_odometry](http://wiki.ros.org/rtabmap_ros#stereo_odometry) approach. See also this [page ](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping#Bring-up_example_with_Zed.2BAC8-Zed-Mini_cameras) for example.

closed 2019-12-20 by chiaueng