introlab/rtabmap #480: ./rtabmap wrong: "Assertion `px != 0' failed. Aborted (core dumped)"

muzizhi, 2019-12-06:
some information:
ubuntu16.04 /and install just by(installation:If ROS is not installed) 
camera is kinect v1, the dirve is openni and sensorkinect; opencv 3.4.3 ;
cmake/make is ok,but when I run: ./rtabmap, I cannot use my camera
something wrong:

`./rtabmap
[ WARN] (2019-12-06 10:45:17.872) Parameters.cpp:1134::readINI() Section "Core" in /home/ocean/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
Program started...
Warning: USB events thread - failed to set priority. This might cause loss of data...
rtabmap: /usr/include/boost/smart_ptr/shared_ptr.hpp:641: typename boost::detail::sp_dereference<T>::type boost::shared_ptr<T>::operator*() const [with T = xn::NodeInfo; typename boost::detail::sp_dereference<T>::type = xn::NodeInfo&]: Assertion `px != 0' failed.
Aborted (core dumped)
`

I find a problem look simlarly [github343:](https://github.com/introlab/rtabmap/issues/343) 
but the most important is that my pcl is 1.7(libpcl-dev) ,and the answer is not for me.
btw,I can show you some more detail:

`gdb ./rtabmap
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./rtabmap...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/ocean/lz/rtabmap/bin/rtabmap 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc69de700 (LWP 28014)]
[New Thread 0x7fffc30cc700 (LWP 28015)]
[New Thread 0x7fffb90f4700 (LWP 28016)]
[New Thread 0x7fffb3fff700 (LWP 28017)]
[New Thread 0x7fffb37fe700 (LWP 28018)]
[New Thread 0x7fffb276d700 (LWP 28019)]
[New Thread 0x7fffb12d6700 (LWP 28021)]
[Thread 0x7fffb276d700 (LWP 28019) exited]
[ WARN] (2019-12-06 09:35:21.800) Parameters.cpp:1134::readINI() Section "Core" in /home/ocean/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
Program started...
[New Thread 0x7fffb276d700 (LWP 28028)]
[New Thread 0x7fffa1bc6700 (LWP 28033)]
Warning: USB events thread - failed to set priority. This might cause loss of data...
[Thread 0x7fffa1bc6700 (LWP 28033) exited]
[New Thread 0x7fffa1bc6700 (LWP 28040)]
[Thread 0x7fffa1bc6700 (LWP 28040) exited]
rtabmap: /usr/include/boost/smart_ptr/shared_ptr.hpp:641: typename boost::detail::sp_dereference<T>::type boost::shared_ptr<T>::operator*() const [with T = xn::NodeInfo; typename boost::detail::sp_dereference<T>::type = xn::NodeInfo&]: Assertion `px != 0' failed.

Thread 1 "rtabmap" received signal SIGABRT, Aborted.
0x00007ffff476d428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb)`

I don't know how to solve it,please give me some advice.

matlabbe, 2019-12-12:
It seems crashing in PCL openni related functions: http://docs.pointclouds.org/1.7.0/structopenni__wrapper_1_1_open_n_i_driver_1_1_device_context.html

Try freenect driver instead. (install `libfreenect-dev` and rebuild rtabmap).

muzizhi, 2019-12-12:
thanks，I will try it later