introlab/rtabmap #483: Error happen when cmake..

aliangboom, 2019-12-16:
I want to install the RTAB-Map in ubuntu 16.04 and download RTAB-Map source. When $ cmake .. ,it say:
(base) jiliangli@jiliang-ThinkStation-P520:~/rtabmap/build$ cmake ..
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /home/jiliangli/anaconda3/include /home/jiliangli/anaconda3/lib/libsqlite3.so
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv4
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.2;/usr/include/x86_64-linux-gnu;/usr/include/jsoncpp;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/python2.7;/usr/include/hdf5/serial;/usr/include/libxml2;/usr/include/tcl
-- Found ZLIB: /home/jiliangli/anaconda3/include
-- VTK_RENDERING_BACKEND=OpenGL
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
CMake Warning at cmake_modules/FindG2O.cmake:122 (MESSAGE):
  Latest g2o version detected with c++11 interface (config file:
  /usr/local/include/g2o/config.h).  Make sure g2o is built with
  "-DBUILD_WITH_MARCH_NATIVE=OFF" to avoid segmentation faults caused by
  Eigen.
Call Stack (most recent call first):
  CMakeLists.txt:374 (FIND_PACKAGE)

-- Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include/eigen3
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found octomap 1.9.0: /usr/local/include
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 2.0.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations]
-- Architecture: x86_64
-- Found Pthreads
-- yaml-cpp not found, euroc_dataset tool won't be built...
-- --------------------------------------------
-- Info :
--   Version : 0.19.6
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DEIGEN_USE_NEW_STDVECTOR;-DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET;-DFLANN_STATIC;-Dqh_QHpointer
--   PCL_VERSION = 1.7.2
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = NO (not found, License: BSD)
[... 181 lines cut ...]
[ 82%] Building CXX object tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/main.cpp.o
/home/jiliangli/anaconda3/lib/libpng16.so.16：‘inflateValidate@ZLIB_1.2.9’
/home/jiliangli/anaconda3/lib/libfontconfig.so.1：‘FT_Done_MM_Var’
/home/jiliangli/anaconda3/lib/libharfbuzz.so.0：‘FT_Get_Var_Blend_Coordinates’
collect2: error: ld returned 1 exit status
tools/StereoEval/CMakeFiles/stereoEval.dir/build.make:450: recipe for target '../bin/rtabmap-stereoEval' failed
make[2]: *** [../bin/rtabmap-stereoEval] Error 1
CMakeFiles/Makefile2:766: recipe for target 'tools/StereoEval/CMakeFiles/stereoEval.dir/all' failed
make[1]: *** [tools/StereoEval/CMakeFiles/stereoEval.dir/all] Error 2
make[1]: *** 
/home/jiliangli/anaconda3/lib/libpng16.so.16：‘inflateValidate@ZLIB_1.2.9’onda3/lib/libfontconfig.so.1：
/home/jiliangli/anaconda3/lib/libharfbuzz.so.0：‘FT_Get_Var_Blend_Coordinates’
collect2: error: ld returned 1 exit status
tools/Camera/CMakeFiles/camera.dir/build.make:450: recipe for target '../bin/rtabmap-camera' failed
make[2]: *** [../bin/rtabmap-camera] Error 1
CMakeFiles/Makefile2:710: recipe for target 'tools/Camera/CMakeFiles/camera.dir/all' failed
make[1]: *** [tools/Camera/CMakeFiles/camera.dir/all] Error 2
[ 82%] Linking CXX executable ../../../bin/rtabmap-kitti_dataset
[ 82%] Built target kitti_dataset
[ 82%] Linking CXX executable ../../../bin/rtabmap-rgbd_dataset
[ 82%] Built target rgbd_dataset
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Please give me some help, thanks.

matlabbe, 2019-12-19:
Remove anaconda paths in your `LD_LIBRARY_PATH` or `PATH` before making rtabmap. The wrong libraries are taken from your anaconda installation instead of the system ones, which may have incompatibilities.

aliangboom, 2019-12-21:
> Remove anaconda paths in your `LD_LIBRARY_PATH` or `PATH` before making rtabmap. The wrong libraries are taken from your anaconda installation instead of the system ones, which may have incompatibilities. Remove `rtabmap/build` directory before retrying to clear cmake cache.

Thanks, it works!

eric-schleicher, 2021-01-22:
I'm having this same issue but with a different twist.  I have a workspace with another package that seems to be causing this issue.  if i stick rtabmap_ros in it's own workspace, no issues...

however with this other ros package in a common workspace.  when i check the LD_LIBRARY_PATH I can see that the that the  lib folder from the workspaces /devel/lib is not in the LD_LIBRARY_PATH.   

This is causing an idential error as above for rtabmap_ros.  

any idea?