introlab/rtabmap #487: Occipital Structure Core Support

promethe42, 2020-01-03:
Hi

I'd like to use rtabmap with the Structure Core:

https://structure.io/structure-core

I'll likely submit a PR for this if/when it works. 
Where should I begin? 

Regards

Keudeugi, 2020-01-06:
I developed roughly working interface with Structure Core (firmware version around 0.7.0?), however I'm afraid if it is suitable to be shared (in terms of code readability, and it is not enough tested). One thing which I can tell you is possible to build. I will post a code if someone wants.

promethe42, 2020-01-06:
@Keudeugi please share your work if you can in a fork or in a PR

matlabbe, 2020-01-06:
Are they OpenNI2 compatible? If so, you could try OpenNI2 driver. 

There are many camera driver examples here: https://github.com/introlab/rtabmap/tree/master/corelib/src/camera

One could call the new class like `CameraStructureCore`. I could help for the integration with the UI, but it could be tested locally with the [No Events C++ example](https://github.com/introlab/rtabmap/wiki/Cplusplus-RGBD-Mapping#example-without-threads-and-events) by changing `CameraStereoImages` by `CameraStructureCore`: https://github.com/introlab/rtabmap/blob/356094fac433f183e935d4315fb5c86aa79ca938/examples/NoEventsExample/main.cpp#L97

Keudeugi, 2020-01-07:
@promethe42, Do you need this immediately? I've checked the code and I found there are a lot of hard coded implementations ( e.g., offline image acquisition; since Structure Core uses its own recording database called OCC ), and it doesn't follow rtabmap's coding styles (I'm using clang-format). If you have some patience, please wait until this February so that I make the codes readable, because I'm very busy with writing a paper for submission.

@matlabbe, I've implemented that by referring those kind of other implementations, and checked that it runs with (real-time)rgbd-mapping example.

Keudeugi, 2020-02-11:
Bump, sorry for the waiting.

Here's bare implementation code for Structure Core interface that I wrote before: https://pastebin.com/0YccThq8
Some features (e.g., integrating IMU data) are not supported so it supports only VO and I didn't made PR, however it'll be greatly appreciated if anyone extend features.