introlab/rtabmap #490: make fail Jetson TX2 JetPack 4.3 Ubuntu 18.04 Ros Melodic

bj-neilson, 2020-01-13:
Hi Mathieu,

With the release of JP4.3 I decided to try again to upgrade to 18.04/Melodic on the TX2.

I've carefully followed the guide for Tegra: https://github.com/introlab/rtabmap_ros 

I can't get past this broken link as below:

ben@ben-tx2:~/rtabmap/build$ make
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libGL.so', needed by '../bin/librtabmap_core.so.0.19.6'.  Stop.
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

Any thoughts?
Thank you,
Ben

matlabbe, 2020-01-13:
Hi,
from the instructions on this page (related to a previous jetpack), you may have to adjust this line:
```bash
$ cd /usr/lib/aarch64-linux-gnu/
$ sudo ln -sf tegra/libGL.so libGL.so
```
Find where `libGL.so` is installed on your computer.

CluelessIT, 2020-02-23:
Hi @matlabbe  

I also encountered this issue when trying to make on my Jetson nano. I found the file libGL.so which is in the directory as stated in the page cd /usr/lib/aarch64-linux-gnu/ . Any other solutions to work out on?

Thank you

tdam2112, 2020-02-28:
Encountered the same issue, running the lines shown by @matlabbe did not change my error.

matlabbe, 2020-03-02:
Note that `/usr/lib/aarch64-linux-gnu/libGL.so` is a symlink to real `libGL.so`. On jetpack3, libGl was located in `/usr/lib/aarch64-linux-gnu/tegra`. On newer jetpack >=4, you may try this:
```bash
$ cd /usr/lib/aarch64-linux-gnu/
$ sudo ln -sf libGL.so.1.0.0 libGL.so
```

tdam2112, 2020-03-04:
> Note that `/usr/lib/aarch64-linux-gnu/libGL.so` is a symlink to real `libGL.so`. On jetpack3, `libGL.so` was located in `/usr/lib/aarch64-linux-gnu/tegra`. On newer jetpack >=4, you may try this:
> 
> ```shell
> $ cd /usr/lib/aarch64-linux-gnu/
> $ sudo ln -sf libGL.so.1.0.0 libGL.so
> ```

This has worked for me, I'm running ubuntu 18.04 on an Jetson Xavier. 
Thanks @matlabbe

CluelessIT, 2020-03-04:
it worked for me too! Thanks for the help @matlabbe! however I met with a new issue this time, where I launch RTABMap standalone but I can't select my intel realsense D435 camera even thou it is connected and the realsense-viewer is able to read the camera. Any idea why it is like this?

bj-neilson, 2020-03-04:
Thanks @matlabbe 

Wish I could have offered advice from my experience but I sort of just try typing a command, close my eyes and pray then press enter.  Sometimes not quite sure which optimistic attempt was the right one 😁
Ben

matlabbe, 2020-03-10:
@CluelessIT When you do `cmake ..`, make sure RealSense2 library is detected by cmake, you should see `With RealSense2           = YES`:
```
$ cd rtabmap/build
$ cmake ..

[...]

--  Camera Drivers:
--   With Freenect             = YES (License: Apache v2 and/or GPLv2)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (WITH_ZED=OFF)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = YES (License: Apache-2)

```

@bj-neilson being optimistic is a great way to solve problems ;)