introlab/rtabmap #491: Installation error

Zumbalamambo, 2020-01-13:
I'm getting the following error message while install rtabmap

> /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libSM.so: undefined reference to `uuid_generate@UUID_1.0'

how do I fix it?

matlabbe, 2020-01-13:
Which Ubuntu version are you using? Which version of libSM is installed?
```bash
$ ls /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libSM.so*
```
Can you also show output of cmake?
```bash
$ cd rtabmap/build
$ cmake ..
```

Zumbalamambo, 2020-01-14:
I'm using Linux Mint 19.

Following is the output of cmake

> -- --------------------------------------------
> -- Info :
> --   Version : 0.19.6
> --   CMAKE_INSTALL_PREFIX = /usr/local
> --   CMAKE_BUILD_TYPE =     Release
> --   CMAKE_INSTALL_LIBDIR = lib
> --   BUILD_APP =            ON
> --   BUILD_TOOLS =          ON
> --   BUILD_EXAMPLES =       ON
> --   BUILD_SHARED_LIBS =    ON
> --   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
> --   FLANN_KDTREE_MEM_OPT = OFF
> --   PCL_DEFINITIONS =  ;-DDISABLE_ENSENSO;-DDISABLE_DAVIDSDK;-DDISABLE_DSSDK;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0;-DFLANN_STATIC;-Dqh_QHpointer;-DDISABLE_RSSDK
> --   PCL_VERSION = 1.8.1
> -- 
> -- Optional dependencies ('*' affects some default parameters) :
> --  *With OpenCV 4 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = NO (not found, License: BSD)
> --   With Qt5                  = YES (License: Open Source or Commercial)
> --   With VTK 6.3              = YES (License: BSD)
> --   With external SQLite3     = YES (License: Public Domain)
> --   With ORB OcTree           = YES (License: GPLv3)
> --   With Madgwick             = YES (License: GPL)
> --   With FastCV               = NO (FastCV not found)
> -- 
> --  Solvers:
> --   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
> --  *With g2o                  = NO (WITH_G2O=OFF)
> --  *With GTSAM                = YES (License: BSD)
> --  *With Ceres                = NO (Ceres not found)
> --   With VERTIGO              = YES (License: GPLv3)
> --   With cvsba                = NO (cvsba not found)
> --  *With libpointmatcher      = NO (libpointmatcher not found)
> -- 
> --  Reconstruction Approaches:
> --   With OCTOMAP              = NO (octomap not found)
> --   With CPUTSDF              = NO (CPUTSDF not found)
> --   With OpenChisel           = NO (open_chisel not found)
> --   With AliceVision          = NO (WITH_ALICE_VISION=OFF)
> -- 
> --  Camera Drivers:
> --   With Freenect             = NO (libfreenect not found)
> --   With OpenNI2              = YES (License: Apache v2)
> --   With Freenect2            = NO (libfreenect2 not found)
> --   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
> --   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
> --   With dc1394               = YES (License: LGPL)
> --   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
> --   With ZED                  = NO (ZED sdk not found)
> --   With RealSense            = NO (librealsense not found)
> --   With RealSense2           = NO (librealsense2 not found)
> -- 
> --  Odometry Approaches:
> --   With loam_velodyne        = NO (loam_velodyne not found)
> --   With libfovis             = NO (libfovis not found)
> --   With libviso2             = NO (libviso2 not found)
> --   With dvo_core             = NO (dvo_core not found)
> --   With okvis                = NO (okvis not found)
> --   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
> --   With VINS-Fusion          = NO (VINS-Fusion not found)
> --   With ORB_SLAM2            = NO (ORB_SLAM2 not found, make sure environment variable ORB_SLAM2_ROOT_DIR is set)
> -- --------------------------------------------
> -- Configuring done

closed 2020-01-14 by Zumbalamambo

edgarcamilocamacho, 2020-07-05:
@Zumbalamambo, did you get solution for it?

kneehit, 2020-07-21:
@edgarcamilocamacho  

Make sure you have uuid-dev library.

Then add following line in CMakeLists.txt fixed the problem for me
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -luuid")`

This fixed the problem for me in Ubuntu 18.04

roalchaq, 2020-11-10:
I had the same issue described here under Ubuntu 20.04. It is solved as described by @kneehit.

```console
/usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libSM.so
/usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libSM.so.6
/usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libSM.so.6.0.1
```