introlab/rtabmap #492: Realsense D435 Missing frames

PhilSc20, 2020-01-14:
Hi there,
I am trying to run rtabmap on Ubuntu 18.04 with a Realsense D435 camera. When I open the Intel realsense viewer, everything works fine and I can see the camera stream. However, if press the start button in rtabmap, an error occurs as follows: 
**[ERROR] (2020-01-14 13:29:18.877) CameraRealSense2.cpp:1209::captureImage() Missing frames (received 1)**

Also, a window pops up, saying: **The camera has reached the end of the stream.**

Thanks for any help!

PhilSc20, 2020-01-15:
Now it seems that the camera stream in rtabmap will work if I just start and stop it for a random amount of times. Sometimes it starts after the third try and sometimes it needs to be started and stopped for over 20 times. I have no idea why that is, as I didn't change anything else.

matlabbe, 2020-01-15:
Which realsense sdk version are you using? Which rtabmap version?

PhilSc20, 2020-01-16:
I am using 
realsense sdk version: 2.31.0
rtabmap version: 0.19.6

matlabbe, 2020-01-17:
I reproduced your problem. It seems to happen randomly. I even reproduced the problem seen in this [post](https://github.com/IntelRealSense/librealsense/issues/4505#issuecomment-572527704) in realsense-viewer where the image stamps are going backward! I am also using a D435i, I got a lot of warnings about imu cannot be interpolated when starting realsense in rtabmap, and sometimes it causes latencies (like 2-3 frames per second). I reproduced also the timestamp not synced between  image and imu data of the original [issue](https://github.com/IntelRealSense/librealsense/issues/5319) in realsense-viewer. Some people seem to say that it may be related to ubuntu kernel version. I'll try with older kernel versions.

Versions I tested:
* rtabmap from source 0.19.6 (latest)
* realsense binaries 2.31.0
* firmware D400: 05.12.01.00

matlabbe, 2020-01-24:
Update: I don't have issues with kernel 4.15, see also this comment: https://github.com/IntelRealSense/librealsense/issues/4505#issuecomment-578250537

closed 2020-01-24 by matlabbe