introlab/rtabmap #494: Memory leak in localization mode?

matlabbe, 2020-01-15:
How reproduce the problem:
 1. Create a database using the [robot mapping demo](http://wiki.ros.org/rtabmap_ros#Robot_mapping). There should be around 228 nodes added to map at the end of the bag and the System Monitor would show around 430 MB usage for rtabmap node.

 2. Using valgrind's [massif](http://valgrind.org/docs/manual/ms-manual.html) tool, reprocess the database offline:
    ````bash
    $ valgrind --tool=massif rtabmap-reprocess rtabmap.db output.db
    ````
 
 3. Opening the resulting massif log in [massif-visualizer](https://github.com/KDE/massif-visualizer), we get (memory is increasing as expected in SLAM mode):
    [image]

 4. Now, redo massif in localization mode:
    ````bash
    $ valgrind --tool=massif --Mem/IncrementalMemory false "rtabmap.db;rtabmap.db" output.db
    ````

 5. See the resulting result:
    [image]

The question is why after initialization at the beginning, the memory is still increasing? It seems everything is constant, except the bottom red area (the colored labels represent the peaks, most of them happening at the beginning).

timonmeyer, 2021-02-09:
Hey @matlabbe,
Is there any update regarding this issue?

matlabbe, 2021-02-11:
I remember having done some memory optimizations and added statistics past in September/October. The results I had didn't show leaks on the experiments I've done though, so it is why I didn't update this issue. 

For this example, I think I've found what is happening. The memory is increasing because proximity detection with laser scans is enabled. On proximity detection, scans of the map are decompressed and **kept not compressed in memory afterwards**. It is also why the final memory usage is mostly the same after the run.

https://github.com/introlab/rtabmap/blob/481a140f845a7488aad9b2805b3997989ad1b6ef/corelib/src/Memory.cpp#L3074-L3080

SLAM ( max = 344.6 MB):
```
valgrind --tool=massif  rtabmap-reprocess rtabmap.db output.db
```
[image]

Localization (with proximity detection enabled, max = 344.6 MB ->exactly the same size than at the end of SLAM phase = **No leaks**):
```
valgrind --tool=massif  rtabmap-reprocess --Mem/IncrementalMemory false "rtabmap.db;rtabmap.db" output.db
```
[image]

To compare, here is localization without proximity detection and without loop closure, to avoid uncompressing the scans. The orange (some sqlite3 calls) is increasing, but that memory is related to DBReader::getNextData() used to load the data to be published for localization.
```
valgrind --tool=massif  rtabmap-reprocess --Mem/IncrementalMemory false --RGBD/ProximityBySpace false --Kp/MaxFeatures -1 "rtabmap.db;rtabmap.db" output.db
```
[image]

Localization (without proximity detection but with loop closure enabled without laser scans). This setup is like doing only visual loop closure detection (in a setup without scans).
```
valgrind --tool=massif  rtabmap-reprocess --Mem/IncrementalMemory false --RGBD/ProximityBySpace false --Reg/Strategy 0 "rtabmap.db;rtabmap.db" output.db
```
[image]

In conclusion, I am not sure why DBReader::getNextData() increases for some time and stop increasing after a while. There is no leak on rtabmap side though.

timonmeyer, 2021-02-11:
Thank you for the very detailed explanation!!

closed 2021-06-09 by matlabbe

ksvbka, 2024-06-07:
Thank for very interesting topic. I can learn alot from it!