introlab/rtabmap #496: Standalone rtabmap crash due to g2o error

m-pilia, 2020-01-22:
I built rtabmap (standalone application, ROS-free) from source (with g2o and GTSAM integration) and tried to use it with a RealSense D435. Whenever the g2o optimiser is invoked (BA or loop closure), rtabmap crashes with the following error (always happens immediately when starting a mapping if g2o is used in the odometry):

```
[FATAL] (2020-01-22 09:23:34.101) OptimizerG2O.cpp:1615::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
```

The problem does not happen with other solvers, e.g. if I replace g2o with ceres or GTSAM in the config for BA and loop closure. I also did not encounter this error when using the binary `.deb` package from the ROS deb repository, so I assume it is not an issue with the camera. (I tried the binary distribution only as a ROS node and not using the standalone rtabmap, because the version distributed in the deb package is not built with `librealsense2` integration).

### How to reproduce
1. Build the dependencies and rtabmap as described [in the wiki](https://github.com/introlab/rtabmap/wiki/Installation#ubuntu) (CMake output in the details section)
2. Start rtabmap standalone GUI (run `./bin/rtabmap` from the root folder of the git sources)
3. Create new database from the toolbar button
4. Set RealSense D435 source from the drop-down menu in the interface
5. Click start

### Result
Crash happens with log message:
```
[FATAL] (2020-01-22 09:23:34.101) OptimizerG2O.cpp:1615::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
```

### Details
* Ubuntu 16.04.6 LTS, kernel 4.15.0
* g2o built from source (fcba4eaca6f20d9a5792404cc8ef303aeb8ba5d2)
* GTSAM 4.0.2 built from source (26a5a42a38375d539162f8094ab273e032e2146c)
* rtabmap 0.19.6 built from source (commit f9e818c900104e4f10c6a2c55494acce000161ee)

<details>
<summary>Output of cmake (click to expand)</summary>

```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/lib/ccache/cc
-- Check for working C compiler: /usr/lib/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/lib/ccache/c++
-- Check for working CXX compiler: /usr/lib/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
[... 168 lines cut ...]
With OpenNI2:            false
With Freenect:            true
With Freenect2:          false
With K4W2:               false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:          true
With libpointmatcher:     true
With octomap:            false
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM2:          false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
```
</details>

matlabbe, 2020-01-31:
g2o is built from source, I generally test with libg2o installed from ros (`ros-kinetic-libg2o` for 16.04), maybe there is a difference. Here what is said on g2o [documentation](https://github.com/RainerKuemmerle/g2o/blob/31ccdcc3bd628edb66db18c7f635d3f368770b77/g2o/core/optimizable_graph.h#L618-L624):
```cpp
    /**
     * verify that all the information of the edges are semi positive definite, i.e.,
     * all Eigenvalues are >= 0.
     * @param verbose output edges with not PSD information matrix on cerr
     * @return true if all edges have PSD information matrix
     */
    bool verifyInformationMatrices(bool verbose = false) const;
```

I tried on an Ubuntu 16.04 computer (kernel 4.4.0) with realsense 2.25.0, D435i and latest g2o built from source. I don't have problems with default parameters with g2o as the graph optimization approach. Loop closures are accepted as expected without crashing. 

It could be an Eigen problem, do you have multiple Eigen libraries installed? If g2o is built against an Eigen version and rtabmap with another it could create bugs like that. You may try building against g2o binaries instead of source to see if it could be a dependency problem.

cheers,
Mathieu

m-pilia, 2020-02-06:
Hi Mathieu, thanks for the reply!

I built all packages against the system version of Eigen, but it may be possible that I made some mistake in the configuration. Will try against a binary version of g2o, since I also had other issues with g2o built from source in other tools.

I would close this issue since I will not be able to follow up. Thanks for the help!

closed 2020-02-06 by m-pilia