introlab/rtabmap #497: Manual Loop Closure?

Marsfan, 2020-01-26:
I am using rtabmap for a personal project to build a 3D model of a with a Kinect (I know that this is not the intended purpose, but it actually works better than any other free software I have found. If you have any advice on best practices for this, I suppose it could be a secondary question). 

However, I have to re-start the odometry at times, as I loose track of the surroundings with my scanner. Sometimes when I do this, the program fails to have a loop closure link to the previous point cloud. Is there a way I can manually tell rtabmap how to link the point clouds together after I finish scanning?

matlabbe, 2020-02-05:
Yes, we can do this with [rtabmap-databaseViewer](https://github.com/introlab/rtabmap/wiki/Tools#database-viewer). Open "View->Constraints view" and "View->Graph view". Using the two main sliders, you can browse images from two different sessions included in the database (i.e., images before and after you reset odometry). When you find two images that should match, try hitting the "Add" button on Constraints view. It may fail to add the loop closure, look at the terminal to see why. You can lower some thresholds by opening "View->Core Parameters" and open the Vis panel.

Marsfan, 2020-02-09:
Oh cool. I somehow missed the add button. 
What I was looking for though was an even more manual way, where I could manually pick points on each image that I know are aligned. Based on your response, I guess rtabmap does not have support for this yet.

matlabbe, 2020-02-20:
Indeed, we cannot pick points ourself like in CloudCompare. I tagged this issue to enhancement.