introlab/rtabmap #498: error: call of overloaded ‘PointXYZRGB(int, int, int)’ is ambiguous

parkerlreed, 2020-01-28:
Arch Linux
rtabmap standalone

```
gcc 9.2.0-4
pcl-git r12100.51d0156a8-1
gtsam-git r11318.4226c12dd-1
g2o-git r1057.fcba4ea-1
libfreenect-git v0.5.7.r29.g106245f-1
libfreenect2-git 0.2.0.r66.gda68a71-1
flann 1.9.1-6
```

Had to add this to the top of the CMakeLists.txt to make PCL happy
```
SET(CMAKE_CXX_STANDARD 14)
SET(CMAKE_CXX_STANDARD_REQUIRED ON)
```

Get this error while building
```
[ 66%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/CloudViewerInteractorStyle.cpp.o
/home/parker/rtabmap/rtabmap/guilib/src/CloudViewerInteractorStyle.cpp: In member function ‘virtual void rtabmap::CloudViewerInteractorStyle::OnMouseMove()’:
/home/parker/rtabmap/rtabmap/guilib/src/CloudViewerInteractorStyle.cpp:133:31: error: call of overloaded ‘PointXYZRGB(int, int, int)’ is ambiguous
  133 |    pcl::PointXYZRGB pt(255,0,0);
      |                               ^
In file included from /usr/include/pcl-1.10/pcl/point_types.h:359,
                 from /usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:41,
                 from /usr/include/pcl-1.10/pcl/common/copy_point.h:58,
                 from /usr/include/pcl-1.10/pcl/common/impl/io.hpp:45,
                 from /usr/include/pcl-1.10/pcl/common/io.h:586,
                 from /usr/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:46,
                 from /usr/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/include/pcl-1.10/pcl/visualization/interactor_style.h:44,
                 from /home/parker/rtabmap/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewerInteractorStyle.h:15,
                 from /home/parker/rtabmap/rtabmap/guilib/src/CloudViewerInteractorStyle.cpp:8:
/usr/include/pcl-1.10/pcl/impl/point_types.hpp:648:12: note: candidate: ‘pcl::PointXYZRGB::PointXYZRGB(float, float, float)’
  648 |     inline PointXYZRGB(float _x, float _y, float _z)
      |            ^~~~~~~~~~~
/usr/include/pcl-1.10/pcl/impl/point_types.hpp:638:12: note: candidate: ‘pcl::PointXYZRGB::PointXYZRGB(uint8_t, uint8_t, uint8_t)’
  638 |     inline PointXYZRGB (std::uint8_t _r, std::uint8_t _g, std::uint8_t _b)
      |            ^~~~~~~~~~~
/home/parker/rtabmap/rtabmap/guilib/src/CloudViewerInteractorStyle.cpp: In member function ‘virtual void rtabmap::CloudViewerInteractorStyle::OnLeftButtonDown()’:
/home/parker/rtabmap/rtabmap/guilib/src/CloudViewerInteractorStyle.cpp:279:31: error: call of overloaded ‘PointXYZRGB(int, int, int)’ is ambiguous
  279 |    pcl::PointXYZRGB pt(255,0,0);
      |                               ^
In file included from /usr/include/pcl-1.10/pcl/point_types.h:359,
                 from /usr/include/pcl-1.10/pcl/common/impl/copy_point.hpp:41,
                 from /usr/include/pcl-1.10/pcl/common/copy_point.h:58,
                 from /usr/include/pcl-1.10/pcl/common/impl/io.hpp:45,
                 from /usr/include/pcl-1.10/pcl/common/io.h:586,
                 from /usr/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:46,
                 from /usr/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/include/pcl-1.10/pcl/visualization/interactor_style.h:44,
                 from /home/parker/rtabmap/rtabmap/guilib/src/../include/rtabmap/gui/CloudViewerInteractorStyle.h:15,
                 from /home/parker/rtabmap/rtabmap/guilib/src/CloudViewerInteractorStyle.cpp:8:
/usr/include/pcl-1.10/pcl/impl/point_types.hpp:648:12: note: candidate: ‘pcl::PointXYZRGB::PointXYZRGB(float, float, float)’
  648 |     inline PointXYZRGB(float _x, float _y, float _z)
      |            ^~~~~~~~~~~
/usr/include/pcl-1.10/pcl/impl/point_types.hpp:638:12: note: candidate: ‘pcl::PointXYZRGB::PointXYZRGB(uint8_t, uint8_t, uint8_t)’
  638 |     inline PointXYZRGB (std::uint8_t _r, std::uint8_t _g, std::uint8_t _b)
      |            ^~~~~~~~~~~
make[2]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/build.make:459: guilib/src/CMakeFiles/rtabmap_gui.dir/CloudViewerInteractorStyle.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:880: guilib/src/CMakeFiles/rtabmap_gui.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

jungladicitta, 2020-02-04:
@parkerlreed Hello! I've got similar issue, have you solved it?

jungladicitta, 2020-02-04:
I just commented that function `inline PointXYZRGB (std::uint8_t _r, std::uint8_t _g, std::uint8_t _b)` in `/usr/local/include/pcl-1.10/pcl/impl/point_types.hpp`
have built successfully

parkerlreed, 2020-02-04:
Thanks. I ended up using an Ubuntu chroot since vtk and others also don't work with the newer versions (And the older vtk6 is a pita to compile)

matlabbe, 2020-02-06:
The commit above should fix the compilation problem. Tested with latest PCL from source.

Thx for reporting!

closed 2020-02-06 by matlabbe

parkerlreed, 2020-02-06:
Thanks! I'll give the "native" compile another shot.

parkerlreed, 2020-02-07:
@matlabbe Just a heads up. This all seems to work fine but I'm hitting an issue on linking with gtsam metis and system metis not getting along. I filed an issue over there first as I'm not sure what camp this falls under. https://github.com/borglab/gtsam/issues/220

matlabbe, 2020-02-20:
I am currently using [4.0.0-alpha2](https://github.com/borglab/gtsam/tags) version of gtsam, if it could help to track the problem. rtabmap doesn't use metis directly, but it is a dependency of gtsam.

commit c9f21e7e (2020-02-06): fixed #498