introlab/rtabmap #499: ZED SDK 3.0 breaks the compilation

lazydroid, 2020-01-29:
Recently (last week) released SDK for ZED cameras from Stereolabs breaks the compilation because of newly introduced types and changes in the namespace. SDK 2.8.* works fine, but does not support the new ZED2 camera.

lazydroid, 2020-01-31:
I have made the changes to ZED related file that allow RTABMap to get compiled with SDK 3.0 and it works with ZED 2. Attaching the file if anyone is interested.

[CameraStereoZed.cpp.zip](https://github.com/introlab/rtabmap/files/4137758/CameraStereoZed.cpp.zip)

matlabbe, 2020-01-31:
See pull request #501 and test it. I'll close this issue in the meantime, feel free to comment if the pull request is not working for you.

closed 2020-01-31 by matlabbe

lazydroid, 2020-02-03:
Your code fails to compile because the arguments of `RuntimeParameters` have changed :

```
rtabmap/corelib/src/camera/CameraStereoZed.cpp:545:121: error: no matching function for call to ‘sl::RuntimeParameters::RuntimeParameters(sl::SENSING_MODE, bool, bool, sl::REFERENCE_FRAME)’
```

matlabbe, 2020-02-03:
The commit above should fix the problem, thx for testing it!

lazydroid, 2020-02-04:
It compiles and works now, thank you very much for your support!!