introlab/rtabmap #502: Fatal error: cs.h not found

masterdot, 2020-02-01:
The file is truly not there and I didnt find anything helping with this.
Any help appreciated :)

`
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv4
-- Found PCL: /usr/include/pcl-1.8;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.3;/usr/include/freetype2;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent/include;/usr/lib/x86_64-linux-gnu/openmpi/include;/usr/include/python2.7;/usr/include/x86_64-linux-gnu;/usr/include/hdf5/openmpi;/usr/include/libxml2;/usr/include/jsoncpp;/usr/include/tcl
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Could NOT find CSPARSE (missing: CSPARSE_INCLUDE_DIR) 
CMake Warning at cmake_modules/FindG2O.cmake:122 (MESSAGE):
  Latest g2o version detected with c++11 interface (config file:
  /usr/local/include/g2o/config.h).  Make sure g2o is built with
  "-DBUILD_WITH_MARCH_NATIVE=OFF" to avoid segmentation faults caused by
  Eigen.
Call Stack (most recent call first):
  CMakeLists.txt:374 (FIND_PACKAGE)

-- Found g2o: /usr/local/include
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/include/eigen3
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Architecture: x86_64
-- Found Pthreads
CMake Deprecation Warning at tools/KittiDataset/CMakeLists.txt:14 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
[... 147 lines cut ...]
linux:~/3dscan/rtabmap/build$ make -j4
[  4%] Built target rtabmap_utilite
[  4%] Built target res_tool
[  5%] Built target imagesJoiner
[  5%] Built target extractObject
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Odometry.cpp.o
[  5%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OdometryThread.cpp.o
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryF2M.cpp.o
In file included from /usr/local/include/g2o/solvers/csparse/csparse_helper.h:30:0,
                 from /usr/local/include/g2o/solvers/csparse/linear_solver_csparse.h:30,
                 from /home/3dscan/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:66:
/usr/local/include/g2o/solvers/csparse/csparse_extension.h:27:10: fatal error: cs.h: Datei oder Verzeichnis nicht gefunden
 #include <cs.h>
          ^~~~~~
compilation terminated.
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1387: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o] Error 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
:~/3dscan/rtabmap/build$ 
`

matlabbe, 2020-02-03:
`cs.h` is located by default here: `/usr/include/suitesparse/cs.h`. Normally the include directory of `cs.h` should be added to compilation: https://github.com/introlab/rtabmap/blob/e2e2227895f84ef9cea238d2805d8bc1d77465f0/cmake_modules/FindG2O.cmake#L2

Make sure `libsuitesparse-dev` is installed:
```bash
sudo apt install libsuitesparse-dev
```

masterdot, 2020-02-04:
Hey there,
that did work and took my to another issue:
`In file included from /usr/local/include/opencv4/opencv2/core/core_c.h:48:0,
                 from /usr/local/include/opencv4/opencv2/imgproc/types_c.h:46,
                 from /usr/local/include/opencv4/opencv2/imgproc/imgproc_c.h:46,
                 from /home/masterdot/3dscan/rtabmap/guilib/src/CalibrationDialog.cpp:33:
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note: candidate: CvSize::CvSize()
 typedef struct CvSize
                ^~~~~~
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note:   candidate expects 0 arguments, 1 provided
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note: candidate: constexpr CvSize::CvSize(const CvSize&)
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note:   no known conversion for argument 1 from ‘cv::Size {aka cv::Size_<int>}’ to ‘const CvSize&’
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note: candidate: constexpr CvSize::CvSize(CvSize&&)
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note:   no known conversion for argument 1 from ‘cv::Size {aka cv::Size_<int>}’ to ‘CvSize&&’
In file included from /home/masterdot/3dscan/rtabmap/guilib/src/CalibrationDialog.cpp:40:0:
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:35: error: ‘c_T’ was not declared in this scope
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                   ^~~
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:35: note: suggested alternative: ‘c_Q’
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                   ^~~
                                   c_Q
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:48: error: ‘c_R2’ was not declared in this scope
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                                ^~~~
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:48: note: suggested alternative: ‘c_R1’
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                                ^~~~
                                                c_R1
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:55: error: ‘c_P1’ was not declared in this scope
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                                       ^~~~
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:55: note: suggested alternative: ‘c_R1’
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                                       ^~~~
                                                       c_R1
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:62: error: ‘c_P2’ was not declared in this scope
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                                              ^~~~
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1214:62: note: suggested alternative: ‘c_Q’
         CvSize(imageSize), &c_R, &c_T, &c_R1, &c_R2, &c_P1, &c_P2, p_Q, flags, alpha,
                                                              ^~~~
                                                              c_Q
/home/masterdot/3dscan/rtabmap/guilib/src/opencv/stereoRectifyFisheye.h:1215:28: error: no matching function for call to ‘CvSize::CvSize(cv::Size&)’
         CvSize(newImageSize));
                            ^
In file included from /usr/local/include/opencv4/opencv2/core/core_c.h:48:0,
                 from /usr/local/include/opencv4/opencv2/imgproc/types_c.h:46,
                 from /usr/local/include/opencv4/opencv2/imgproc/imgproc_c.h:46,
                 from /home/masterdot/3dscan/rtabmap/guilib/src/CalibrationDialog.cpp:33:
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note: candidate: CvSize::CvSize()
 typedef struct CvSize
                ^~~~~~
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note:   candidate expects 0 arguments, 1 provided
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note: candidate: constexpr CvSize::CvSize(const CvSize&)
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note:   no known conversion for argument 1 from ‘cv::Size {aka cv::Size_<int>}’ to ‘const CvSize&’
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note: candidate: constexpr CvSize::CvSize(CvSize&&)
/usr/local/include/opencv4/opencv2/core/types_c.h:1169:16: note:   no known conversion for argument 1 from ‘cv::Size {aka cv::Size_<int>}’ to ‘CvSize&&’
[ 66%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/PostProcessingDialog.cpp.o
guilib/src/CMakeFiles/rtabmap_gui.dir/build.make:720: recipe for target 'guilib/src/CMakeFiles/rtabmap_gui.dir/CalibrationDialog.cpp.o' failed
make[2]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/CalibrationDialog.cpp.o] Error 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
/home/masterdot/3dscan/rtabmap/guilib/src/ExportBundlerDialog.cpp: In member function ‘void rtabmap::ExportBundlerDialog::exportBundler(const std::map<int, rtabmap::Transform>&, const std::multimap<int, rtabmap:: [... cut ...]
/home/masterdot/3dscan/rtabmap/guilib/src/ExportBundlerDialog.cpp:251:24: warning: ‘*((void*)&<anonymous> +560)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
        if(velocity.size() == 6)
           ~~~~~~~~~~~~~^~
/home/masterdot/3dscan/rtabmap/guilib/src/ExportBundlerDialog.cpp:251:24: warning: ‘*((void*)&<anonymous> +568)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
CMakeFiles/Makefile2:341: recipe for target 'guilib/src/CMakeFiles/rtabmap_gui.dir/all' failed
make[1]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
`

matlabbe, 2020-02-05:
I updated the code to build with latest OpenCV version from source. See commit above.

masterdot, 2020-02-05:
Ok, done with that one :)

Heres another one:
`collect2: error: ld returned 1 exit status
tools/KittiDataset/CMakeFiles/kitti_dataset.dir/build.make:470: recipe for target '../bin/rtabmap-kitti_dataset' failed
make[2]: *** [../bin/rtabmap-kitti_dataset] Error 1
CMakeFiles/Makefile2:822: recipe for target 'tools/KittiDataset/CMakeFiles/kitti_dataset.dir/all' failed
make[1]: *** [tools/KittiDataset/CMakeFiles/kitti_dataset.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
`

Sorry for the problems...

matlabbe, 2020-02-05:
It says there is an error in the generated Makefiles, but not acutally saying what is the error... Try cleanup the build directory and remake.
```bash
rm -rf rtabmap/build/*
cd rtabmap/build
cmake ..
make -j4
```

masterdot, 2020-02-05:
I'm sorry to bother you again... Last time I tried to clear the build folder didn't work, but this time it did. Sorry for this beginner's error -. -

The next one is this:
`[ 70%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_MainWindow.cpp.o
[ 71%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_PreferencesDialog.cpp.o
[ 71%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DatabaseViewer.cpp.o
[ 71%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_AboutDialog.cpp.o
[ 72%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ConsoleWidget.cpp.o
[ 72%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ImageView.cpp.o
[ 73%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_PdfPlot.cpp.o
[ 73%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_StatsToolBox.cpp.o
[ 73%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ProgressDialog.cpp.o
[ 74%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/utilite/moc_UPlot.cpp.o
[ 74%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/utilite/moc_UImageView.cpp.o
[ 74%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CloudViewer.cpp.o
[ 75%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_OdometryViewer.cpp.o
[ 75%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_LoopClosureViewer.cpp.o
[ 75%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DataRecorder.cpp.o
[ 76%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CameraViewer.cpp.o
[ 76%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CalibrationDialog.cpp.o
[ 76%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ExportDialog.cpp.o
[ 77%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_PostProcessingDialog.cpp.o
[ 77%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ExportCloudsDialog.cpp.o
[ 77%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ExportBundlerDialog.cpp.o
[ 78%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_MapVisibilityWidget.cpp.o
[ 78%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_GraphViewer.cpp.o
[ 78%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_CreateSimpleCalibrationDialog.cpp.o
[ 79%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_ParametersToolBox.cpp.o
[ 79%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DepthCalibrationDialog.cpp.o
[ 79%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_EditConstraintDialog.cpp.o
[ 80%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/3rdParty/moc_QMultiComboBox.cpp.o
[ 80%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_TexturingState.cpp.o
[ 80%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_RecoveryState.cpp.o
[ 81%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_EditDepthArea.cpp.o
[ 81%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_EditMapArea.cpp.o
[ 81%] Building CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/qrc_GuiLib.cpp.o
[ 82%] Linking CXX shared library ../../../bin/librtabmap_gui.so
[ 82%] Built target rtabmap_gui
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
masterdot@masterdot-linux:~/3dscan/rtabmap/build$ `

It comes out of nowhere... Seems to be a issue with the gui building, but I don't find a hint what might be wrong... Maybe a lib is missing, but I don't know which may be... 
Thanks for your patience and help!

matlabbe, 2020-02-05:
Use "make" (without -j4), so only file is built at the same time to make easier to read the error.

masterdot, 2020-02-06:
OK, quick update: I did recompile g2o and found out that it does use the EIGEN package. The result is: YES! It did compile without errors and I've successfully build it now.
You should add the requirement of the EIGEN fro the main package and the g2o. 

Hope I could help you, thanks a lot for helping. Hope I do actually use it :)

matlabbe, 2020-02-06:
RTAB-Map doesn't depend on Eigen directly, it is why it is not explicitly specified (when building against PCL, the eigen headers directory is automatically included in rtabmap compilation). It looks more a g2o error to me, not sure eigen was the problem, unless g2o was using another version than the one used by PCL. Great that you managed to build it! 

cheers,
Mathieu

closed 2020-02-06 by matlabbe