introlab/rtabmap #503: DLT Algorithm cv::Exception

masterdot, 2020-02-06:
`terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0-dev) /home/masterdot/3dscan/opencv/modules/calib3d/src/calibration.cpp:1173: error: (-2:Unspecified error) in function 'void cvFindExtrinsicCameraParams2(const CvMat*, const CvMat*, const CvMat*, const CvMat*, CvMat*, CvMat*, int)'
> DLT algorithm needs at least 6 points for pose estimation from 3D-2D point correspondences. (expected: 'count >= 6'), where
>     'count' is 5
> must be greater than or equal to
>     '6' is 6

Abgebrochen (Speicherabzug geschrieben)
`

matlabbe, 2020-02-06:
See https://answers.ros.org/question/330555/cannot-localize-using-rtabmap/?answer=330942#post-id-330942

Three months ago, I added this to avoid this problem:
https://github.com/introlab/rtabmap/blob/179f17d2671f48472a9ee860165873ea06db16c3/corelib/src/RegistrationVis.cpp#L122-L126

Do you have the latest version? If so, we would need to know which function is rtabmap is calling this. Try launching rtabmap in gdb:
$ sudo apt install gdb
$ gdb rtabmap

Type "start" in gdb console. When it crashes, type "bt" and copy paste the log here.

masterdot, 2020-02-09:
Hmm... Currently I can't reconstruct the error. It scans without any errors and this particular error doesnt happen right now... Close for a while, I might reopen it, when its happening again...

commit 034adb9f (2020-04-02): fixed #503
closed 2020-02-09 by masterdot