introlab/rtabmap #507: g2o fatal error

mikuia09, 2020-02-12:
Hi!
I built rtabmap (standalone version)  from source as what wiki writes, with g2o and GTSAM built from source too.
When trying to play dataset from ICL-NUIM, the program crashed and the description is :
<div style="background: #ffffff; overflow:auto;width:auto;border:solid gray;border-width:.1em .1em .1em .8em;padding:.2em .6em;"><pre style="margin: 0; line-height: 125%">OptimizerG2O<span style="color: #333333">.</span>cpp:<span style="color: #0000DD; font-weight: bold">1615</span>::optimizaeBA()Condition
(optimizer<span style="color: #333333">.</span>verifyinformationMatrices())<span style="color: #000000; font-weight: bold">not</span> met<span style="color: #FF0000; background-color: #FFAAAA">!</span>
</pre></div>
I setted the source as RGB-D and data from images. I'm not sure whtether it is caused by the camera calibration matrix of the dataset. In ICL-NUIM dataset, the fy is negative. However, in rtabmap, when I try to create calibration file, the fy must be positive. If I change the number to negative in the file, it seems the program can't recognize the calibration file and points out that I must calibrate the camera before using RGB-D SLAM.
My system is ubuntu 16.04.
Hope for help.
Thanks

matlabbe, 2020-02-21:
Which g2o version are you using? is it the latest from [master branch](https://github.com/RainerKuemmerle/g2o)? Is PCL also built from source?

mikuia09, 2020-03-10:
> Which g2o version are you using? is it the latest from [master branch](https://github.com/RainerKuemmerle/g2o)? Is PCL also built from source?

Thanks for your replay.
I gitted clone the g2o souce code recently, and built it. PCL is not built from source, I used apt-get to install pcl lib about 2 years ago.

matlabbe, 2020-03-10:
Can you try with this release? https://github.com/RainerKuemmerle/g2o/releases/tag/20170730_git
just to see if it is g2o the problem (that version should work).

mikuia09, 2020-03-30:
> Can you try with this release? https://github.com/RainerKuemmerle/g2o/releases/tag/20170730_git
> just to see if it is g2o the problem (that version should work).

It works!
Thanks for your answer! I'll close the issue.

closed 2020-03-30 by mikuia09