introlab/rtabmap #509: mutex isnt getting locked here

katrinbschmid, 2020-02-18:
hi, My subclassed driver fails here in a multi threading run and i noticed the mutex isnt getting locked here. I checked the history and its been like that for a while. Is that intentional? Thanks for your help

https://github.com/introlab/rtabmap/blob/ab9f05fb0ffe70062b1d2becaa587eb3f131a210/corelib/src/DBDriver.cpp#L297

void DBDriver::commit() const
{
	ULOGGER_DEBUG("");
	this->executeNoResultQuery("COMMIT;");
	_transactionMutex.unlock();
}

katrinbschmid, 2020-02-18:
Ah its looked in this->beginTransaction(); i wonder why it would corrupt  data there anyways

closed 2020-02-19 by katrinbschmid

matlabbe, 2020-02-21:
`beginTransaction()` and `commit()` should be used together like here:
https://github.com/introlab/rtabmap/blob/5fef44ec2bd394330f7aa7322837bd20788d8695/corelib/src/DBDriver.cpp#L339-L373

Set this to false to avoid the threading stuff, which could make it easier to debug:
https://github.com/introlab/rtabmap/blob/5fef44ec2bd394330f7aa7322837bd20788d8695/corelib/src/Memory.cpp#L2096