introlab/rtabmap #511: Using RTABmap as a server on PC to a mobile device?

crogersarvr, 2020-02-26:
I saw there was some sort of Tango streamer (my tango phone is broken :-( ).

I would like to run RTABlab on my laptop (ideally my jetson nano but im having compilation problems related to cvbsa) and have it serve map data that I will view on an Oculus Quest.
Is there options or a code example to do this?
thanks

matlabbe, 2020-03-02:
cvbsa is not required, you can opt-out the option with cmake : 
```bash
cd rtabmap/build
cmake -DWITH_CVSBA=OFF ..
```
Do you want to do SLAM on server (PC) and transmit **online** the 3D clouds to a mobile device? Or just visualize in VR a 3D map created from RTAB-Map? For the later, you could just export the cloud/mesh with RTAB-Map desktop (File->Export 3D clouds...) in a common format (like PLY or OBJ). You may find an VR app able to open this kind of 3D file to visualize it.

chrisrogers3d, 2020-03-15:
im looking to do the slam and pointclouds on the nano and visualize on an oculus quest that has the rgb-d camera attached to it.  i did get rtapmap to work on my jetson, then i ran into the opengl/qt issue so i couldnt visualize on jetson nano yet, i will keep trying (crogersarvr).

matlabbe, 2020-04-02:
For Jetson's Qt/vtk issues, see this post: https://github.com/introlab/rtabmap/issues/427#issuecomment-516914817