introlab/rtabmap #517: Exporting rtabmap coordinates to Unity

ArvindSanjeev, 2020-03-18:
Hello,
I have a setup where I am using two realsense cameras (d435, t265) and launching rtabmap using: roslaunch realsense2_camera rovy_rtabmap.launch mapping:=true

I am able to view the generated map and coordinates in rviz, now I am planning to read the published coordinates through Unity ([ROS sharp](https://github.com/siemens/ros-sharp/tree/master/ROS) ). And I simply want to move the camera in unity based on the real world camera coordinates (position and rotation). I am curious as to which published topics from rtabmap will be the right one for this? Is it any of these:
/rtabmap/mapGraph, /rtabmap/mapGraph/mapToOdom, /rtabmap/mapGraph/poses
or /rtabmap/mapData? Which would be the right one to get the real world camera coordinates?

I also saw in some forums about this topic: /rtabmap/odom, but I don't see it when I do rostopic list.

matlabbe, 2020-04-02:
Hello,

It seems I saw the email you sent me before this one. I'll recopy my answer if someone else has the same problem.

> The t265 relasense2_camera node is posting odometry topic at 30 Hz (or 60 Hz? don't remember). RTAB-Map on its side does SLAM at 1 Hz by default, publishing the correction of odometry at 1 Hz. With rtabmapviz for example, I can subscribe to both odometry from T265 (30 Hz) and rtabmap/mapGraph or rtabmap/mapData to get the correction mapToOdom at 1 Hz.
> 
> It is also possible to convert /tf topic directly, which would contain the same transforms as above. TF is quite standard in ROS, I didn't look deeply in ROS#, but they may be already supporting that topic.

cheers,
Mathieu

closed 2020-04-02 by matlabbe

ArvindSanjeev, 2020-04-05:
Thanks @matlabbe I subscribed directly to the appropriate tf topic.