introlab/rtabmap #518: Unable to activate realsense2 on RTAB-Map standalone

makihiro, 2020-03-19:
I'm trying to use Realsense D435i camera on RTAB-Map standalone app.
However, it's unable to select realsense2 for source.
librealsense2 driver has been installed.  I also tried to build app from the source while checking  "With RealSense2: YES"  option exists in cmake.
Is there some confirming point else?

[image]

matlabbe, 2020-04-02:
Click on ? -> About, and see if `With Realsense2` is also yes or not. Look at the rtabmap version, make sure it is the right rtabmap exec you are starting. For example, if rtabmap ros melodic binaries are also installed, the rtabmap version included in `/opt/ros/melodic/bin` may be used when you type only `$ rtabmap` in terminal.

You can do also `$ which rtabmap`.

makihiro, 2020-04-03:
I made sure that right rtabmap has started. However, I found 'With Realsense2 : No' on the About window. I also found that  'Found Realsense2:  ' path is blank when running cmake.

makihiro, 2020-04-03:
I found realsense2 should be RealSense2 in CMakeLists.txt so that cmake can use cmake_modules/FindRealSense.cmake
https://github.com/introlab/rtabmap/blob/034adb9f1f2de9fa8ff3ee6661e22bbfed15fbe2/CMakeLists.txt#L447

matlabbe, 2020-04-03:
Under linux we don't use the included [FindRealSense2.cmake](https://github.com/introlab/rtabmap/blob/master/cmake_modules/FindRealSense2.cmake) file, we search for "realsense2" to find the provided cmake config file by realsense directly, which would be installed here if binaries are installed:
`/usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Config.cmake`
The resulting library is a target, not the library file path. Here what looks like that file:

```cmake
####### Expanded from @PACKAGE_INIT@ by configure_package_config_file() #######
####### Any changes to this file will be overwritten by the next CMake run ####
####### The input file was realsense2Config.cmake.in                            ########

get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../../../" ABSOLUTE)

macro(set_and_check _var _file)
  set(${_var} "${_file}")
  if(NOT EXISTS "${_file}")
    message(FATAL_ERROR "File or directory ${_file} referenced by variable ${_var} does not exist !")
  endif()
endmacro()

macro(check_required_components _NAME)
  foreach(comp ${${_NAME}_FIND_COMPONENTS})
    if(NOT ${_NAME}_${comp}_FOUND)
      if(${_NAME}_FIND_REQUIRED_${comp})
        set(${_NAME}_FOUND FALSE)
      endif()
    endif()
  endforeach()
endmacro()

####################################################################################

set(realsense2_VERSION_MAJOR "2")
set(realsense2_VERSION_MINOR "33")
set(realsense2_VERSION_PATCH "1")

set(realsense2_VERSION ${realsense2_VERSION_MAJOR}.${realsense2_VERSION_MINOR}.${realsense2_VERSION_PATCH})

set_and_check(realsense2_INCLUDE_DIR "${PACKAGE_PREFIX_DIR}/include")

include("${CMAKE_CURRENT_LIST_DIR}/realsense2Targets.cmake")
set(realsense2_LIBRARY realsense2::realsense2)
```
Which realsense version are you using? In case they changed how they write the file name.

ffrivera0, 2020-09-02:
Hi, I have the same issue when using realsense camera via rtabmap on ubuntu20.04. I followed the instruction [librealsense](https://github.com/IntelRealSense/librealsense) to build librealsense and installed rtabmap followed your instruction. However, I failed to activate realsense using `cmake ..`. Here is the result:
```
-- Info :
--   Version : 0.20.3
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++14
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DDISABLE_OPENNI;-DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0
--   PCL_VERSION = 1.10.0
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.4.0 xfeatures2d = NO, nonfree = NO (License: BSD)
--   With Qt 5.12.8            = YES (License: Open Source or Commercial)
--   With VTK 7.1              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With SupertPoint          = NO (libtorch not found)
--   With Python3              = NO (WITH_PYMATCHER=OFF)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
-- 
--  Solvers:
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o                  = YES (License: BSD)
--  *With GTSAM                = YES (License: BSD)
--  *With Ceres                = YES (License: BSD)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (cvsba not found)
--  *With libpointmatcher      = NO (libpointmatcher not found)
-- 
--  Reconstruction Approaches:
--   With OCTOMAP              = NO (octomap not found)
--   With CPUTSDF              = NO (CPUTSDF not found)
--   With OpenChisel           = NO (open_chisel not found)
--   With AliceVision          = NO (WITH_ALICE_VISION=OFF)
-- 
--  Camera Drivers:
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (VINS-Fusion not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
```
How can I solve the problem? Thank you

matlabbe, 2020-09-02:
Where is `realsense2Config.cmake` installed when building from source? in `/usr/local/cmake/realsense`?

ffrivera0, 2020-09-03:
> Where is `realsense2Config.cmake` installed when building from source? in `/usr/local/cmake/realsense`?

It seems that it is the build error from librealsense and librealsense is failed to install. The issue is still in progress [here](https://github.com/microsoft/vcpkg/issues/11752).

peta-peta, 2020-10-19:
I have similar problems as @makihiro. I cannot get  rtabmap standalone compiled with realsense2. I compiled librealsense and linked all libraries via cmake-gui, because it cannot find realsense2 on its own. But still not working.

[image]

matlabbe, 2020-10-19:
What is the output of the cmake log?

peta-peta, 2020-10-20:
Please find attached the cmake log files from /rtabmap/build/CMakeFiles
[CMakeOutput.log](https://github.com/introlab/rtabmap/files/5406850/CMakeOutput.log)
[CMakeError.log](https://github.com/introlab/rtabmap/files/5406854/CMakeError.log)

Cheers Peter

matlabbe, 2020-10-20:
Well, checking this thread: https://github.com/DecodeGenetics/graphtyper/issues/9
What is the output of 
```bash
cd rtabmap/build
cmake ..
```
? cmake itself is failing, what is your system?

peta-peta, 2020-10-20:
I use Ubuntu 18.04 LTS.
I just ran cmake .. again. Please find attached the log file. Here, i cannot see any error.
[log.txt](https://github.com/introlab/rtabmap/files/5410075/log.txt)

matlabbe, 2020-10-20:
It does detect `--   With RealSense2           = YES (License: Apache-2)`. If you do make, do you have errors?

peta-peta, 2020-10-20:
Yes, it does detect it, but shouldn't there be a Path here:
`-- Found RealSense2: `
I did not have any error while compiling. 

So far I was sourcing Ros-melodic at the same time. I have uncommented it in the .bashrc. Now I am recompiling. I let you know once it is finished.

matlabbe, 2020-10-20:
It is because the new realsense2Config.cmake uses a cmake target instead of using library and include paths. It is correct that the path doesn't appear.

peta-peta, 2020-10-20:
It is working! Apparently ros-melodic caused problems. After using a new shell without sourcing ros-melodic, I recompiled rtabmap. Realsense2 is working now.
Thanks again for your support :)

CeccAnd, 2020-11-25:
Hello I have the same issue. Ubuntu 16 + Realsense2 with T265.

With RealSense2 = YES. 

l tried CMakeLists.txt changes as mentioned above by Makihiro. But with no results. 

Are there other solutions?

matlabbe, 2020-11-26:
Do you have ros-kinetic-rtabmap installed? It seems you are not starting the rtabmap you just built.

CeccAnd, 2020-11-26:
Yes it is /opt/ros/kinetic/bin/rtabmap

CeccAnd, 2020-11-26:
If I use ~/rtabmap/build$ **cmake -DWITH_VINS=OFF ..** I see: Checking for module 'libopenni2' and Could NOT find OpenNI2 and vtkRenderingPythonTkWidgets" references the file "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist. Is something related to Realsense2?

CeccAnd, 2020-11-26:
I did again make and here the result:
[ 80%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_process_frame"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_sensor_info"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_delete_stream_profiles_list"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_extrinsics"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_failed_function"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_stream_to_string"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_query_sensors"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_delete_context"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_stream_profile_data"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_delete_frame_queue"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_is_sensor_extendable_to"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_create_context"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_is_stream_profile_default"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_supports_option"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_start_processing"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_query_devices"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_create_device"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_create_sensor"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_stream_profile_is"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_failed_args"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_load_json"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_depth_scale"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_frame_timestamp"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_stream_profile"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_close"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_stream_profiles"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_create_frame_queue"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_set_devices_changed_callback_cpp"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_delete_device"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_pose_frame_get_pose_data"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_frame_stream_profile"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_stream_profiles_count"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_free_error"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_poll_for_frame"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_error_message"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_embedded_frames_count"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_release_frame"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_frame_add_ref"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_delete_processing_block"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_option"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_enqueue_frame"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_device_info"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_delete_sensor_list"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_delete_sensor"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_frame_data"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_device_count"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_open_multiple"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_is_frame_extendable_to"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_video_stream_resolution"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_stop"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_hardware_reset"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_device_list_contains"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_extract_frame"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_sensors_count"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_is_device_extendable_to"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_keep_frame"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_get_librealsense_exception_type"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_format_to_string"
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_create_sync_processing_block"
[... cut ...]
../../../bin/librtabmap_core.so.0.20.6: riferimento non definito a "rs2_start_cpp"
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:525: set di istruzioni per l'obiettivo "../bin/rtabmap" non riuscito
make[2]: *** [../bin/rtabmap] Errore 1
CMakeFiles/Makefile2:416: set di istruzioni per l'obiettivo "app/src/CMakeFiles/rtabmap.dir/all" non riuscito
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Errore 2
Makefile:149: set di istruzioni per l'obiettivo "all" non riuscito
make: *** [all] Errore 2

matlabbe, 2020-11-30:
I had this error recently, from a PCL built with a previous realsense2 version. Rebuilt PCL without realsense2 or with the same used than rtabmap fixed the problem.

To make sure you start the right rtabmap, remove the binaries:
```
sudo apt remove ros-kinetic-rtabmap
```

sebinefrancis, 2022-04-28:
For me this was a conflict between rtabmap, which I installed using `sudo apt install ros-<ros version name>-rtabmap-ros`) and standalone ubuntu installed rtabmap (`/usr/local/bin/rtabmap`: which i built from source using `cmake .. and make install`). 

So initially I tried to uninstall ros installed version of rtabmap. Then I used which rtabmap in terminal and it showed /usr/local/bin/rtabmap. But when i run it, it says some '... .so' dependencies missing (`error while loading shared libraries: librtabmap_gui.so.0.20: cannot open shared object file: No such file or directory`). 

Hence I needed to keep ros installed version and standalone version together. But should run standalone version when i type rtabmap in terminal. 

So I edited bashrc file (`sudo gedit bashrc`) and commented (add # to begining of line) following: 
#source /opt/ros/<your ros version name>/setup.bash: This is only a workaround 
#source /home/<your user name>/catkin_ws/devel/setup.bash: disable adding catkin work space to bash 
#export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/ros/<your ros version name>/lib/x86_64-linux-gnu 

save close, reopen terminal run rtabmap. Now I get realsense2 enabled. 

So, I needed to keep ros version to provide dependencies but needed to prevent it from loading to bash terminal by commenting out.bashrc, inorder to make standalone rtabmap to open up.

This is only a workaround and not a real solution. But someone get a real solution from this hint.

matlabbe, 2022-05-09:
If you want to use latest stanalone and rtabmap_ros, 1) remove rtabmap and rtabmap_ros ros binaries. 2) build/install rtabmap from source in /usr/local, then 3) clone rtabmap_ros in your catkin workspace. You won't ned to change any environment variables.