introlab/rtabmap #522: rtabmap1.9.3 compiling errors with microsoft visual stuido 2017 on windows 10

zengletian1491, 2020-03-24:
Hello, I compile rtabmap1.9.3 source code with microsoft visual stuido 2017 on Windows 10 system.
However, when I build solution, errors occurred as follows. I don't know how to solve this problem. Could you please give me some suggstions?
[errors.docx](https://github.com/introlab/rtabmap/files/4372864/errors.docx)

zengletian1491, 2020-03-26:
I find the solution：
1）open RTABMap.sln with visual studio 2017
2) add the include directory, the lib directory of GTSAM to the property of the project "rtabmap core"
3) add the additional dependency library of GTSAM
Build (Ctrl + Shift + B)

matlabbe, 2020-04-02:
CMake should have added GTSAM libraries to linker in visual studio solution. Hopefully you've found a solution. I'll close it.

closed 2020-04-02 by matlabbe

jmxntg, 2020-06-07:
Did you manage to build rtabmap windows version? I'm desperately trying to build a version that supports orbbec astra sensor without success. Can you help me please?

matlabbe, 2020-06-23:
Is orbbec astra sensor using a specific driver or is it using OpenNI2? If OpenNI2 is used, this build would support it: https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts