introlab/rtabmap #527: how to study rtabmap code?

fangcywangermu, 2020-04-01:
I am a graduate, and my Graduation thesis is about this with orbslam2, but I cant understand the code,could you please help me?

matlabbe, 2020-04-03:
The general approach studied with this paper: 

> M. Labbé and F. Michaud, “[RTAB-Map as an Open-Source Lidar and Visual SLAM Library for Large-Scale and Long-Term Online Operation](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf),” in Journal of Field Robotics, vol. 36, no. 2, pp. 416–446, 2019.  ([Wiley](https://doi.org/10.1002/rob.21831))

There are also other papers on the [main page](http://introlab.github.io/rtabmap/) specific to other aspects of RTAB-Map, like the memory management approach. For understanding the code, a starting point could be to start by the main function [Rtabmap::process()](https://github.com/introlab/rtabmap/blob/9c62a11316838d0661d4a5751362e2fd07e08adf/corelib/src/Rtabmap.cpp#L956) called for each frame.

cheers,
Mathieu

fangcywangermu, 2020-04-04:
Can you give any suggestions for RTABMAP positioning by combining ORBSLAM2 with lidar?

fangcywangermu, 2020-04-04:
> 可以使用本文研究通用方法：
> 
> > [M.Labbé](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf)和F. Michaud，“ [RTAB-Map作为用于大规模和长期在线操作的开源激光雷达和Visual SLAM库](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf)，”《现场机器人学报》，第1卷。36号 2，pp。416–446，2019.（[Wiley](https://doi.org/10.1002/rob.21831)）
> 
> 也有在其他文件[主页](http://introlab.github.io/rtabmap/)特定RTAB地图的其他方面，如内存管理方法和环路闭合检测。为了理解代码，一个起点可以是从针对每个帧调用的主要函数[Rtabmap :: process（）开始](https://github.com/introlab/rtabmap/blob/9c62a11316838d0661d4a5751362e2fd07e08adf/corelib/src/Rtabmap.cpp#L956)。
> 
> 干杯，
> 马修

Can you give any suggestions for RTABMAP positioning by combining ORBSLAM2 with lidar?

TheBricktop, 2020-07-29:
@fangcywangermu orbslam2 is mainly visual based slam and it needs to be backed by sensor that has image output (ie webcam, stereo cam, rgbd module), by using lidar you probably have in mind using  the rotary mounted plane scanning version, and there are other implementation that use this specific approach like intelligent vacuum cleaners or grass movers. https://github.com/simondlevy/BreezySLAM