introlab/rtabmap #528: Is there any plans to support orbbec cameras?

vangogh-ken, 2020-04-07:
Orbbec cameras becomes to be popular. Is there any plans to support orbbec cameras?

matlabbe, 2020-04-08:
I think you can use OpenNI2 driver to use them in rtabmap. Not sure the [openni download](https://orbbec3d.com/develop/) on their page is compatible with rtabmap, someone could give a try. For the Astra SDK, as I don't own that kind of camera, it would be difficult for me to add support. If someone want to try implementing a new `CameraAstra` class, there are many examples here for other drivers: https://github.com/introlab/rtabmap/tree/master/corelib/src/camera

vangogh-ken, 2020-04-09:
@matlabbe Got it. Thx.

closed 2020-04-09 by vangogh-ken

henriquenunez, 2022-08-03:
Hello! I have an orbbec camera and I want to create support to it.

matlabbe, 2022-11-26:
As mentioned in comment above, OpenNI2 driver would work with orbbec camera, unless they have new cameras with a new API to access their cameras.

henriquenunez, 2022-12-05:
Yes, but one camera specifically requires the Astra SDk, which is the Orbbec Astra S U3 camera. If you plug in more than 1 camera it fails.

matlabbe, 2022-12-10:
Unfortunately, I cannot test Astra SDK, my Astra camera seems only compatible with OpenNI2.