introlab/rtabmap #529: make error with octomap

HaowenLai, 2020-04-13:
Hi, when I am compiling, the "undefined reference" error has occurred:  
  
[ 63%] Linking CXX executable ../../../bin/rtabmap-camera  
../../../bin/librtabmap_core.so.0.19.6：对‘rtabmap::OctoMap::addToCache(int, cv::Mat const&, cv::Mat const&, cv::Mat const&, cv::Point3_<float> const&)' undefined reference  
../../../bin/librtabmap_core.so.0.19.6：对‘rtabmap::OctoMap::createCloud(unsigned int, std::vector<int, std::allocator<int> >*, std::vector<int, std::allocator<int> >*, std::vector<int, std::allocator<int> >*, bool, std::vector<int, std::allocator<int> >*, std::vector<double, std::allocator<double> >*) const’undefined reference  
../../../bin/librtabmap_core.so.0.19.6：对‘rtabmap::OctoMap::~OctoMap()’undefined reference  
../../../bin/librtabmap_core.so.0.19.6：对‘rtabmap::OctoMap::OctoMap(float, float, bool, float)’undefined reference  
../../../bin/librtabmap_core.so.0.19.6：对‘rtabmap::OctoMap::update(std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::Transform> > > const&)’undefined reference  
collect2: error: ld returned 1 exit status
tools/Camera/CMakeFiles/camera.dir/build.make:431: recipe for target '../bin/rtabmap-camera' failed
make[2]: *** [../bin/rtabmap-camera] Error 1
CMakeFiles/Makefile2:710: recipe for target 'tools/Camera/CMakeFiles/camera.dir/all' failed
  
And the same error occurs in linking rtabmap-stereoEval and rtabmap-console. It seems that it is related to librtabmap_core.so.0.19.6. The cmake output is as follows:  
  
Checking for module 'libopenni2'
  No package 'libopenni2' found
Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

Checking for module 'libopenni2'
  No package 'libopenni2' found
Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
Found OpenMP
Found OpenCV: /usr/local/include;/usr/local/include/opencv
Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-6.2;/usr/include/hdf5/openmpi;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent;/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include;/usr/lib/openmpi/include;/usr/lib/openmpi/include/openmpi;/usr/include/libxml2;/usr/include/x86_64-linux-gnu;/usr/include/freetype2;/usr/include/x86_64-linux-gnu/freetype2;/usr/include/python2.7;/usr/include/jsoncpp;/usr/include/tcl
Found ZLIB: /usr/include
VTK_RENDERING_BACKEND=OpenGL
Found Freenect: /usr/include
Found DC1394: /usr/include/dc1394
Old g2o version detected with c++03 interface (config file: /usr/local/include/g2o/config.h).
Found g2o: /usr/local/include;/usr/include/suitesparse;/usr/include/suitesparse
GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include
FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
Found RealSense2: 
Found octomap : /opt/ros/kinetic/include
Found installed version of Eigen: /usr/lib/cmake/eigen3
Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
Found required Ceres dependency: glog
Found Ceres version: 1.14.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, OpenMP, Multithreading]
Architecture: x86_64
Found Pthreads

Info :
  Version : 0.19.6
  CMAKE_INSTALL_PREFIX = /home/derek/catkin_ws/devel
  CMAKE_BUILD_TYPE =     Release
  CMAKE_INSTALL_LIBDIR = lib
  BUILD_APP =            ON
  BUILD_TOOLS =          ON
  BUILD_EXAMPLES =       OFF
  BUILD_SHARED_LIBS =    ON
  CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++11
  FLANN_KDTREE_MEM_OPT = OFF
  PCL_DEFINITIONS = -DEIGEN_USE_NEW_STDVECTOR;-DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET;-DFLANN_STATIC;-Dqh_QHpointer
  PCL_VERSION = 1.7.2

Optional dependencies ('*' affects some default parameters) :
 *With OpenCV 3 xfeatures2d module (SIFT/SURF/BRIEF/FREAK) = YES (License: Non commercial)
  With Qt5                  = YES (License: Open Source or Commercial)
  With VTK 6.2              = YES (License: BSD)
  With external SQLite3     = YES (License: Public Domain)
  With ORB OcTree           = YES (License: GPLv3)
  With Madgwick             = YES (License: GPL)
  With FastCV               = NO (FastCV not found)

 Solvers:
  With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
 *With g2o                  = YES (License: BSD)
 *With GTSAM                = YES (License: BSD)
 *With Ceres                = YES (License: BSD)
  With VERTIGO              = YES (License: GPLv3)
  With cvsba                = NO (cvsba not found)
 *With libpointmatcher      = NO (libpointmatcher not found)

 Reconstruction Approaches:
  With OCTOMAP              = YES (License: BSD)
  With CPUTSDF              = NO (CPUTSDF not found)
  With OpenChisel           = NO (open_chisel not found)
  With AliceVision          = NO (WITH_ALICE_VISION=OFF)

 Camera Drivers:
  With Freenect             = YES (License: Apache v2 and/or GPLv2)
  With OpenNI2              = NO (OpenNI2 not found)
  With Freenect2            = NO (libfreenect2 not found)
  With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
  With Kinect for Azure     = NO (Kinect for Azure SDK not found)
  With dc1394               = YES (License: LGPL)
  With FlyCapture2/Triclops = NO (Point Grey SDK not found)
  With ZED                  = NO (ZED sdk not found)
  With RealSense            = NO (librealsense not found)
  With RealSense2           = YES (License: Apache-2)

 Odometry Approaches:
  With loam_velodyne        = NO (loam_velodyne not found)
  With libfovis             = NO (libfovis not found)
  With libviso2             = NO (libviso2 not found)
  With dvo_core             = NO (dvo_core not found)
  With okvis                = NO (okvis not found)
  With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
  With VINS-Fusion          = NO (VINS-Fusion not found)
  With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
  
I would appreciate your help. Thanks.

matlabbe, 2020-04-17:
This is strange, those functions should be included in librtabmap_core.so. For example, OctoMap::~OctoMap() is defined [here](https://github.com/introlab/rtabmap/blob/de4b17332df3fba67ea8b9c3cb040aadaba67d30/corelib/src/OctoMap.cpp#L329-L333). An example of build can be seen on the buildfarm for xenial: 
https://travis-ci.org/github/introlab/rtabmap/jobs/676374161
You may try to spot the differences.

HaowenLai, 2020-04-18:
Problem solved.  
Previously in my system, octomap library cannot be found with CMake sentence `find_package(OCTOMAP QUIET)` in /rtabmap/CMakeLists.txt, so I change it to `find_package(octomap QUIET)`. And I replace all `OCTOMAP_FOUND` with `octomap_FOUND`. It turned out that I forgot to change it in /rtabmap/corelib/src/CMakeLists.txt, making the file `OctoMap.cpp` not compiling.  
Thank you very much. @matlabbe

closed 2020-04-18 by HaowenLai

matlabbe, 2020-04-18:
I don't know why it was OCTOMAP instead of octomap, though it worked in all my builds. Which cmake version are you using? Maybe cmake is now more case sensitive. I still replaced OCTOMAP by octomap in the commit above.

cheers,
Mathieu

HaowenLai, 2020-04-20:
My CMake version is 3.5.1.  
Thanks.