introlab/rtabmap #530: Google earth

Quitino, 2020-04-15:
Thank you very much for your excellent work. When I read the [wiki](https://github.com/introlab/rtabmap/wiki/Robust-Graph-Optimization), I found that SLAM Trajectory can be imported into Google earth display. How do you do this?thank you.

matlabbe, 2020-04-17:
The database should have GPS recorded in it, then the options:
1.  `File->Export Poses...-> Google Earth (*.kml)`
2. `File->Export GPS...-> Google Earth (*.kml)` 

will be enabled in rtabmap-databaseViewer tool. Example from database downloaded from [here](https://github.com/introlab/rtabmap/wiki/Robust-Graph-Optimization#multi-session):
```
$ rtabmap-databaseViewer loop_3it_gps.db
```
[image]

petterhs, 2021-09-06:
Hi!
Does this method still work? When I try with the example database linked above I get this result:
[image]

Similar results when I use my own DB

matlabbe, 2021-09-08:
Hi,
I tested on https://earth.google.com/web/ an it still work.
[image]
[image]

The kml:
```kml
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2">
<Document>
	<name>/home/mathieu/Downloads/gps.kml</name>
	<StyleMap id="msn_ylw-pushpin">
		<Pair>
			<key>normal</key>
			<styleUrl>#sn_ylw-pushpin</styleUrl>
		</Pair>
		<Pair>
			<key>highlight</key>
			<styleUrl>#sh_ylw-pushpin</styleUrl>
		</Pair>
	</StyleMap>
	<Style id="sh_ylw-pushpin">
		<IconStyle>
			<scale>1.2</scale>
		</IconStyle>
		<LineStyle>
			<color>ff808000</color>
		</LineStyle>
	</Style>
	<Style id="sn_ylw-pushpin">
		<LineStyle>
			<color>ff808000</color>
		</LineStyle>
	</Style>
	<Placemark>
		<name>/home/mathieu/Downloads/gps</name>
		<styleUrl>#msn_ylw-pushpin</styleUrl>		<LineString>
			<coordinates>
				-71.942607,45.378918,174.000000 -71.942605,45.378917,174.000000 -71.942603,45.378916,175.000000 -71.942600,45.378918,175.000000 -71.942599,45.378919,175.000000 -71.942597,45.378920,175.000000 -71.942594,45.378921,176.000000 -71.942592,45.378923,176.000000 -71.942590,45.378924,176.000000 -71.942590,45.378925,176.000000 -71.942586,45.378932,176.000000 -71.942584,45.378936,177.000000 -71.942583,45.378938,177.000000 -71.942586,45.378938,176.000000 -71.942588,45.378940,176.000000 -71.942595,45.378935,175.000000 -71.942595,45.378936,176.000000 -71.942590,45.378941,175.000000 -71.942581,45.378943,175.000000 -71.942571,45.378944,178.000000 -71.942559,45.378945,178.000000 -71.942553,45.378940,179.000000 -71.942545,45.378936,180.000000 -71.942539,45.378933,181.000000 -71.942534,45.378929,181.000000 -71.942526,45.378925,183.000000 -71.942517,45.378919,183.000000 -71.942509,45.378913,180.000000 -71.942500,45.378907,180.000000 -71.942493,45.378901,180.000000 -71.942488,45.378897,179.000000 -71.942483,45.378898,178.000000 -71.942476,45.378903,179.000000 -71.942468,45.378905,179.000000 -71.942464,45.378905,180.000000 -71.942458,45.378905,181.000000 -71.942448,45.378907,181.000000 -71.942440,45.378912,184.000000 -71.942430,45.378915,186.000000 -71.942420,45.378909,190.000000 -71.942411,45.378903,190.000000 -71.942400,45.378896,191.000000 -71.942388,45.378892,193.000000 -71.942384,45.378889,192.000000 -71.942377,45.378887,193.000000 -71.942367,45.378886,194.000000 -71.942357,45.378887,194.000000 -71.942345,45.378886,196.000000 -71.942331,45.378878,197.000000 -71.942325,45.378868,196.000000 -71.942317,45.378861,197.000000 -71.942295,45.378856,198.000000 -71.942272,45.378856,198.000000 -71.942250,45.378867,194.000000 -71.942227,45.378885,196.000000 -71.942208,45.378903,193.000000 -71.942191,45.378919,192.000000 -71.942175,45.378932,192.000000 -71.942147,45.378956,192.000000 -71.942133,45.378969,191.000000 -71.942120,45.378981,192.000000 -71.942107,45.378991,191.000000 -71.942093,45.379000,192.000000 -71.942077,45.379008,191.000000 -71.942061,45.379015,194.000000 -71.942046,45.379023,193.000000 -71.942031,45.379030,192.000000 -71.942018,45.379040,191.000000 -71.942008,45.379047,189.000000 -71.942000,45.379055,189.000000 -71.941991,45.379063,188.000000 -71.941984,45.379071,186.000000 -71.941981,45.379080,186.000000 -71.941979,45.379088,185.000000 -71.941979,45.379097,184.000000 -71.941980,45.379107,183.000000 -71.941981,45.379116,183.000000 -71.941981,45.379135,183.000000 -71.941981,45.379144,181.000000 -71.941981,45.379151,181.000000 -71.941981,45.379160,181.000000 -71.941978,45.379167,181.000000 -71.941974,45.379170,179.000000 -71.941969,45.379175,177.000000 -71.941963,45.379182,178.000000 -71.941955,45.379192,178.000000 -71.941948,45.379200,177.000000 -71.941942,45.379208,176.000000 -71.941937,45.379214,176.000000 -71.941934,45.379222,176.000000 -71.941933,45.379231,177.000000 -71.941931,45.379239,177.000000 -71.941929,45.379247,177.000000 -71.941926,45.379255,177.000000 -71.941923,45.379265,177.000000 -71.941920,45.379275,177.000000 -71.941917,45.379286,175.000000 -71.941915,45.379296,176.000000 -71.941911,45.379302,176.000000 -71.941907,45.379309,177.000000 -71.941905,45.379318,174.000000 -71.941905,45.379320,173.000000 -71.941907,45.379323,173.000000 -71.941913,45.379326,172.000000 -71.941924,45.379333,172.000000 -71.941938,45.379339,172.000000 -71.941951,45.379346,172.000000 -71.941963,45.379353,173.000000 -71.941972,45.379361,174.000000 -71.941980,45.379368,174.000000 -71.941988,45.379373,174.000000 -71.941995,45.379379,176.000000 -71.942014,45.379389,174.000000 -71.942021,45.379396,175.000000 -71.942027,45.379405,175.000000 -71.942034,45.379420,176.000000 -71.942040,45.379432,176.000000 -71.942045,45.379447,177.000000 -71.942049,45.379462,178.000000 -71.942051,45.379474,176.000000 -71.942052,45.379483,176.000000 -71.942051,45.379490,178.000000 -71.942051,45.379494,178.000000 -71.942053,45.379491,176.000000 -71.942054,45.379486,175.000000 -71.942053,45.379481,175.000000 -71.942061,45.379480,175.000000 -71.942075,45.379480,174.000000 -71.942086,45.379479,174.000000 -71.942095,45.379478,174.000000 -71.942097,45.379478,174.000000 -71.942095,45.379477,174.000000 -71.942095,45.379475,174.000000 -71.942092,45.379473,174.000000 -71.942088,45.379473,174.000000 -71.942083,45.379473,174.000000 -71.942080,45.379471,174.000000 -71.942080,45.379471,174.000000 -71.942081,45.379473,174.000000 -71.942081,45.379472,174.000000 -71.942081,45.379471,174.000000 -71.942080,45.379470,174.000000 -71.942079,45.379469,174.000000 -71.942079,45.379470,174.000000 -71.942078,45.379470,174.000000 -71.942078,45.379470,174.000000 -71.94207 [... cut ...] 1,177.000000 -71.942627,45.378959,177.000000 -71.942627,45.378958,177.000000 -71.942627,45.378957,176.000000 -71.942627,45.378958,177.000000 -71.942627,45.378958,177.000000 -71.942627,45.378958,177.000000 -71.942627,45.378958,177.000000 -71.942627,45.378958,177.000000 -71.942627,45.378958,177.000000 -71.942627,45.378958,177.000000 -71.942626,45.378958,177.000000 -71.942626,45.378958,177.000000 -71.942626,45.378958,177.000000 
			</coordinates>
		</LineString>
	</Placemark>
</Document>
</kml>
```

petterhs, 2021-09-13:
Thank you! Found the issue. For some reason the decimal separators are "," in the .kml files I export instead of ".". Probably because commas are defaults in my country. Not sure where to fix it though or if it is a problem elsewhere. 

Should I open a new issue on this?

Here is my kml file:
```
<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2">
<Document>
	<name>/home/petter/Downloads/gps2.kml</name>
	<StyleMap id="msn_ylw-pushpin">
		<Pair>
			<key>normal</key>
			<styleUrl>#sn_ylw-pushpin</styleUrl>
		</Pair>
		<Pair>
			<key>highlight</key>
			<styleUrl>#sh_ylw-pushpin</styleUrl>
		</Pair>
	</StyleMap>
	<Style id="sh_ylw-pushpin">
		<IconStyle>
			<scale>1.2</scale>
		</IconStyle>
		<LineStyle>
			<color>ff808000</color>
		</LineStyle>
	</Style>
	<Style id="sn_ylw-pushpin">
		<LineStyle>
			<color>ff808000</color>
		</LineStyle>
	</Style>
	<Placemark>
		<name>/home/petter/Downloads/gps2</name>
		<styleUrl>#msn_ylw-pushpin</styleUrl>		<LineString>
			<coordinates>
				-71,942607,45,378918,174,000000 -71,942605,45,378917,174,000000 -71,942603,45,378916,175,000000 -71,942600,45,378918,175,000000 -71,942599,45,378919,175,000000 -71,942597,45,378920,175,000000 -71,942594,45,378921,176,000000 -71,942592,45,378923,176,000000 -71,942590,45,378924,176,000000 -71,942590,45,378925,176,000000 -71,942586,45,378932,176,000000 -71,942584,45,378936,177,000000 -71,942583,45,378938,177,000000 -71,942586,45,378938,176,000000 -71,942588,45,378940,176,000000 -71,942595,45,378935,175,000000 -71,942595,45,378936,176,000000 -71,942590,45,378941,175,000000 -71,942581,45,378943,175,000000 -71,942571,45,378944,178,000000 -71,942559,45,378945,178,000000 -71,942553,45,378940,179,000000 -71,942545,45,378936,180,000000 -71,942539,45,378933,181,000000 -71,942534,45,378929,181,000000 -71,942526,45,378925,183,000000 -71,942517,45,378919,183,000000 -71,942509,45,378913,180,000000 -71,942500,45,378907,180,000000 -71,942493,45,378901,180,000000 -71,942488,45,378897,179,000000 -71,942483,45,378898,178,000000 -71,942476,45,378903,179,000000 -71,942468,45,378905,179,000000 -71,942464,45,378905,180,000000 -71,942458,45,378905,181,000000 -71,942448,45,378907,181,000000 -71,942440,45,378912,184,000000 -71,942430,45,378915,186,000000 -71,942420,45,378909,190,000000 -71,942411,45,378903,190,000000 -71,942400,45,378896,191,000000 -71,942388,45,378892,193,000000 -71,942384,45,378889,192,000000 -71,942377,45,378887,193,000000 -71,942367,45,378886,194,000000 -71,942357,45,378887,194,000000 -71,942345,45,378886,196,000000 -71,942331,45,378878,197,000000 -71,942325,45,378868,196,000000 -71,942317,45,378861,197,000000 -71,942295,45,378856,198,000000 -71,942272,45,378856,198,000000 -71,942250,45,378867,194,000000 -71,942227,45,378885,196,000000 -71,942208,45,378903,193,000000 -71,942191,45,378919,192,000000 -71,942175,45,378932,192,000000 -71,942147,45,378956,192,000000 -71,942133,45,378969,191,000000 -71,942120,45,378981,192,000000 -71,942107,45,378991,191,000000 -71,942093,45,379000,192,000000 -71,942077,45,379008,191,000000 -71,942061,45,379015,194,000000 -71,942046,45,379023,193,000000 -71,942031,45,379030,192,000000 -71,942018,45,379040,191,000000 -71,942008,45,379047,189,000000 -71,942000,45,379055,189,000000 -71,941991,45,379063,188,000000 -71,941984,45,379071,186,000000 -71,941981,45,379080,186,000000 -71,941979,45,379088,185,000000 -71,941979,45,379097,184,000000 -71,941980,45,379107,183,000000 -71,941981,45,379116,183,000000 -71,941981,45,379135,183,000000 -71,941981,45,379144,181,000000 -71,941981,45,379151,181,000000 -71,941981,45,379160,181,000000 -71,941978,45,379167,181,000000 -71,941974,45,379170,179,000000 -71,941969,45,379175,177,000000 -71,941963,45,379182,178,000000 -71,941955,45,379192,178,000000 -71,941948,45,379200,177,000000 -71,941942,45,379208,176,000000 -71,941937,45,379214,176,000000 -71,941934,45,379222,176,000000 -71,941933,45,379231,177,000000 -71,941931,45,379239,177,000000 -71,941929,45,379247,177,000000 -71,941926,45,379255,177,000000 -71,941923,45,379265,177,000000 -71,941920,45,379275,177,000000 -71,941917,45,379286,175,000000 -71,941915,45,379296,176,000000 -71,941911,45,379302,176,000000 -71,941907,45,379309,177,000000 -71,941905,45,379318,174,000000 -71,941905,45,379320,173,000000 -71,941907,45,379323,173,000000 -71,941913,45,379326,172,000000 -71,941924,45,379333,172,000000 -71,941938,45,379339,172,000000 -71,941951,45,379346,172,000000 -71,941963,45,379353,173,000000 -71,941972,45,379361,174,000000 -71,941980,45,379368,174,000000 -71,941988,45,379373,174,000000 -71,941995,45,379379,176,000000 -71,942014,45,379389,174,000000 -71,942021,45,379396,175,000000 -71,942027,45,379405,175,000000 -71,942034,45,379420,176,000000 -71,942040,45,379432,176,000000 -71,942045,45,379447,177,000000 -71,942049,45,379462,178,000000 -71,942051,45,379474,176,000000 -71,942052,45,379483,176,000000 -71,942051,45,379490,178,000000 -71,942051,45,379494,178,000000 -71,942053,45,379491,176,000000 -71,942054,45,379486,175,000000 -71,942053,45,379481,175,000000 -71,942061,45,379480,175,000000 -71,942075,45,379480,174,000000 -71,942086,45,379479,174,000000 -71,942095,45,379478,174,000000 -71,942097,45,379478,174,000000 -71,942095,45,379477,174,000000 -71,942095,45,379475,174,000000 -71,942092,45,379473,174,000000 -71,942088,45,379473,174,000000 -71,942083,45,379473,174,000000 -71,942080,45,379471,174,000000 -71,942080,45,379471,174,000000 -71,942081,45,379473,174,000000 -71,942081,45,379472,174,000000  [... cut ...] 1,177,000000 -71,942627,45,378959,177,000000 -71,942627,45,378958,177,000000 -71,942627,45,378957,176,000000 -71,942627,45,378958,177,000000 -71,942627,45,378958,177,000000 -71,942627,45,378958,177,000000 -71,942627,45,378958,177,000000 -71,942627,45,378958,177,000000 -71,942627,45,378958,177,000000 -71,942627,45,378958,177,000000 -71,942626,45,378958,177,000000 -71,942626,45,378958,177,000000 -71,942626,45,378958,177,000000 
			</coordinates>
		</LineString>
	</Placemark>
</Document>
</kml>
```

matlabbe, 2021-09-13:
Hi, 

This is a common problem for any code using "printf" or "fprintf" with `%f` format. Depending on your computer locale, it will output a `.`or a `,`.  The values are exported here:
https://github.com/introlab/rtabmap/blob/2fcef88016b90c5e10b1b1eb012b76cfd3fc84f1/corelib/src/Graph.cpp#L526

Related issue: http://official-rtab-map-forum.206.s1.nabble.com/play-rgbd-dataset-freiburg2-tp1714p1787.html
```
$ export LANG="en_US.UTF-8"
```
see also https://stackoverflow.com/questions/12845638/how-do-i-change-the-decimal-separator-in-the-printf-command-in-bash

matlabbe, 2025-02-02:
Fixed in https://github.com/introlab/rtabmap/commit/6d713c2ccbda3d511050b053522a76a003430f1d

closed 2025-02-02 by matlabbe