introlab/rtabmap #533: Virtual RGBD and Rtabmap : TF troubles

tomlogan501, 2020-04-22:
Hi,
I got an issue when I launch this command 
`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rgb_topic:=/zed_cam1/image_raw  depth_topic:=/zed/depth/image_raw camera_info_topic:=/zed_cam1/camera_info frame_id:=base_link approx_sync:=true queue_size:=30`

Rtabmap seems to not get the orientation from the virtual ZED.

The scene in Gazebo 
[image]

The tf 
[image]

Rtabviz
[image]

Shoud I also turn the frames of the rgb cam with rpy(-PI/2 , 0 , -PI/2)?

tomlogan501, 2020-04-22:
Seems that it occurs after I launch rtabmap.
Before it looks like this
[image]

Is there a tf manipulation inside rtabmap on the depth sensor ?

matlabbe, 2020-04-27:
You should set `frame_id` to `base_footprint`:

`roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rgb_topic:=/zed_cam1/image_raw depth_topic:=/zed/depth/image_raw camera_info_topic:=/zed_cam1/camera_info frame_id:=base_link approx_sync:=true queue_size:=30 frame_id:=base_footprint`

Also what is the frame set in zed topics?
```$ rostopic echo /zed_cam1/camera_info/header/frame_id
$ rostopic echo /zed/depth/image_raw/header/frame_id
$ rostopic echo/zed_cam1/image_raw/header/frame_id```

tomlogan501, 2020-04-27:
$ rostopic echo /zed_cam1/camera_info/header/frame_id
-->zed_cam1_optical
$ rostopic echo /zed/depth/image_raw/header/frame_id
-->zed_optical_frame
$ rostopic echo/zed_cam1/image_raw/header/frame_id
-->zed_cam1_optical

The zed_optical_frame has a rpy(-pi/2,0,-pi/2) from zed_frame due to the plugin gazebo_ros_camera_depth

I didn't test the real ZED for now, I just create a virtual one based on the datasheet to emulate some behavior on our main robot.

matlabbe, 2020-04-27:
What is the difference between zed_optical_frame and zed_cam1_optical? There should have indeed an optical rotation (-pi/2,0,-pi/2) between the frame of the camera topics and base_link.

tomlogan501, 2020-04-28:
zed_cam1_optical represent the frame for the video output of the first camera
zed_optical_frame (not a good name) is the frame given to the camera depth plugin with the correct optical translation.
If I rotate zed_cam1_optical, it won't correspond to the real ZED I think.

matlabbe, 2020-04-29:
Just checked on my zedm, the depth, color and camera_info topics are all on `zedm_left_camera_optical_frame` frame, for which the Tf from `base_link` is:
```
rosrun tf tf_echo base_link zedm_left_camera_optical_frame
At time 0.000
- Translation: [0.000, 0.030, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000]
```