introlab/rtabmap #537: issue with installation on Mac OS

3DMiller, 2020-04-24:
When downloading the dependencies via Macports through the command line, the error appears:  "Error: Failed to configure libpcl: vtk must be installed with +qt5" . Any ideas on what should be done? I am a beginner user, so not a lot of experience running things in the Terminal.

matlabbe, 2020-04-27:
You may try:
```
sudo port install vtk +qt5
```

3DMiller, 2020-04-29:
Hi, 

I think that worked. I’m sorry for the question being very elementary, but if I already downloaded RTAB Map from the dmg file, do I need to run the commands in the terminal? Or, will RTAB Map know where to locate the dependencies to use them? I think qt% worked and I downloaded the rest through Macports. Is the installation complete?

Thanks,

Spencer Miller
Masters student in Marine Science - Hawaiʻi Pacific University 
www.millerspencer.weebly.com <http://www.millerspencer.weebly.com/>
www.sketchfab.com/S_Miller <http://www.sketchfab.com/S_Miller> 
https://jmadinlab.github.io/people/spencer <https://jmadinlab.github.io/people/spencer> 

> On Apr 26, 2020, at 3:38 PM, matlabbe <notifications@github.com> wrote:
> 
> sudo port install vtk +qt5

matlabbe, 2020-05-26:
The dmg contains already all libraries to run in standalone. I am currently testing building rtabmap on Mac with latest ports from MacPort. See https://github.com/introlab/rtabmap/issues/543

closed 2020-05-26 by matlabbe