introlab/rtabmap #538: Build failed

ankitvora7, 2020-04-25:
Hi I was trying to build RTABMAP using the tutorial and I was unable to build it from source. Here's the error log
```
[ 18%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Parameters.cpp.o
[ 18%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Signature.cpp.o
[ 19%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Features2d.cpp.o
[ 19%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Transform.cpp.o
[ 19%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/GeodeticCoords.cpp.o
[ 20%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util2d.cpp.o
[ 20%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d.cpp.o
/home/akic/code/common/rtabmap/corelib/src/Features2d.cpp: In member function 'virtual void rtabmap::SIFT::parseParameters(const ParametersMap&)':
/home/akic/code/common/rtabmap/corelib/src/Features2d.cpp:844:19: error: incomplete type 'CV_SIFT {aka cv::xfeatures2d::SIFT}' used in nested name specifier
  _sift = CV_SIFT::create(this->getMaxFeatures(), nOctaveLayers_, contrastThreshold_, edgeThreshold_, sigma_);
                   ^~~~~~
/home/akic/code/common/rtabmap/corelib/src/Features2d.cpp: In member function 'virtual std::vector<cv::KeyPoint> rtabmap::SIFT::generateKeypointsImpl(const cv::Mat&, const Rect&, const cv::Mat&)':
/home/akic/code/common/rtabmap/corelib/src/Features2d.cpp:862:7: error: invalid use of incomplete type 'class cv::xfeatures2d::SIFT'
  _sift->detect(imgRoi, keypoints, maskRoi); // Opencv keypoints
       ^~
In file included from /home/akic/code/common/rtabmap/corelib/src/Features2d.cpp:28:0:
/home/akic/code/common/rtabmap/corelib/src/../include/rtabmap/core/Features2d.h:65:7: note: forward declaration of 'class cv::xfeatures2d::SIFT'
 class SIFT;
       ^~~~
/home/akic/code/common/rtabmap/corelib/src/Features2d.cpp: In member function 'virtual cv::Mat rtabmap::SIFT::generateDescriptorsImpl(const cv::Mat&, std::vector<cv::KeyPoint>&) const':
/home/akic/code/common/rtabmap/corelib/src/Features2d.cpp:874:7: error: invalid use of incomplete type 'class cv::xfeatures2d::SIFT'
  _sift->compute(image, keypoints, descriptors);
       ^~
In file included from /home/akic/code/common/rtabmap/corelib/src/Features2d.cpp:28:0:
/home/akic/code/common/rtabmap/corelib/src/../include/rtabmap/core/Features2d.h:65:7: note: forward declaration of 'class cv::xfeatures2d::SIFT'
 class SIFT;
       ^~~~
[ 20%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_filtering.cpp.o
[ 21%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/util3d_mapping.cpp.o
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:883: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Features2d.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Features2d.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:59:0:
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp: In instantiation of 'void pcl::ExtractIndices<PointT>::filterDirectly(pcl::ExtractIndices<PointT>::PointCloudPtr&) [with PointT = pcl::PointNormal; pcl::ExtractIndices<PointT>::PointCloudPtr = boost::shared_ptr<pcl::PointCloud<pcl::PointNormal> >]':
/home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:64:1:   required from here
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp:62:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (pt_index >= input_->points.size ())
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp: In instantiation of 'void pcl::ExtractIndices<PointT>::applyFilter(pcl::ExtractIndices<PointT>::PointCloud&) [with PointT = pcl::PointNormal; pcl::ExtractIndices<PointT>::PointCloud = pcl::PointCloud<pcl::PointNormal>]':
/home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:64:1:   required from here
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp:95:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       if (pt_index >= input_->points.size ())
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52:0,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/segmentation/sac_segmentation.h:49,
                 from /home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:44:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function 'void __static_initialization_and_destruction_0(int, int)':
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: 'pcl::SAC_SAMPLE_SIZE' is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
In file included from /home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:59:0:
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp: In instantiation of 'void pcl::ExtractIndices<PointT>::applyFilter(pcl::ExtractIndices<PointT>::PointCloud&) [with PointT = pcl::PointXYZRGBNormal; pcl::ExtractIndices<PointT>::PointCloud = pcl::PointCloud<pcl::PointXYZRGBNormal>]':
/home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:1803:1:   required from here
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp:95:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       if (pt_index >= input_->points.size ())
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp: In instantiation of 'void pcl::ExtractIndices<PointT>::applyFilter(pcl::ExtractIndices<PointT>::PointCloud&) [with PointT = pcl::PointXYZRGB; pcl::ExtractIndices<PointT>::PointCloud = pcl::PointCloud<pcl::PointXYZRGB>]':
/home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:1803:1:   required from here
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp:95:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp: In instantiation of 'void pcl::ExtractIndices<PointT>::applyFilter(pcl::ExtractIndices<PointT>::PointCloud&) [with PointT = pcl::PointXYZ; pcl::ExtractIndices<PointT>::PointCloud = pcl::PointCloud<pcl::PointXYZ>]':
/home/akic/code/common/rtabmap/corelib/src/util3d_filtering.cpp:1803:1:   required from here
/usr/include/pcl-1.8/pcl/filters/impl/extract_indices.hpp:95:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```
Before trying to install RTABMAP, I installed opencv (master branch) with opencv_contrib (master branch). I also installed g20 (master branch) from source with ``` DBUILD_WITH_MARCH_NATIVE=OFF ``` flag. Post that I installed GTSAM (master branch) with ``` GTSAM_USE_SYSTEM_EIGEN=ON ``` flag. I tried following the tutorial but seems like I messed up somewhere. Can you assist ? Thanks

matlabbe, 2020-04-27:
It seems opencv_contrib master is broken, as SIFT [has been moved](https://github.com/opencv/opencv_contrib/commit/1d77931b30b26c40b60c4fef931d8f0f048dc6ba) from nonfree to free, but only appears in 4.3.0 branch (SIFT just doesn't exist in master branch). Try rebuild opencv with opencv_contrib 4.3.0:
```
cd opencv_contrib
git checkout 4.3.0
... go make opencv
```

Zumbalamambo, 2020-05-08:
+1  same error though contrrib is installed with opencv 4.3

matlabbe, 2020-05-09:
SIFT is in xfeatures2d.hpp for 4.3.0: https://github.com/opencv/opencv_contrib/blob/4.3.0/modules/xfeatures2d/include/opencv2/xfeatures2d.hpp#L74

Verify that in your installed xfeatures2d.hpp that SIFT is there.

matlabbe, 2020-05-09:
I've found the problem. No need to use 4.3.0 from opencv_contrib, they actually moved SIFT back to original features2d.hpp in opencv git. I fixed the problem with the commit above.

closed 2020-05-09 by matlabbe

tkircher, 2020-05-09:
Yeah, SIFT is out of patent now.

phil333, 2020-06-11:
having the same problem with the latest version of opencv 3.4. Manually fixing typedef cv::xfeatures2d::SIFT CV_SIFT;

commit abc3ebee (2020-05-09): fixed #538