introlab/rtabmap #541: Nano make failing during rtabmap build "error: ‘cvZero’ was not declared in this scope"

tmichals, 2020-05-03:
I am trying to build rtabmap on my Nano....
[image]

but I am getting the following error:

[image]

It looks like rtabmap will not build with OpenCV.    Nano Jetson comes with OpenCV4?

lcs@lcs:~$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'4.1.1'

[rtabmap make Failing on OpenCV.txt](https://github.com/introlab/rtabmap/files/4571953/rtabmap.make.Failing.on.OpenCV.txt)

tkircher, 2020-05-03:
Try adding

#include <opencv2/core/core_c.h>

to rtabmap/corelib/src/StereoCameraModel.cpp

tmichals, 2020-05-04:
Thank you!  That worked!   
Should I have known core_c.h was missing?   I knew the error was in StereoCameraModel.cpp.

closed 2020-05-04 by tmichals

tkircher, 2020-05-04:
The transition from OpenCV 3 to OpenCV 4 has been long and complicated. Many people are having these kinds of issues. I'm sure it will all get worked out by OpenCV 5. ;)

tkircher, 2020-05-04:
@matlabbe This might be something that needs an #ifdef.

matlabbe, 2020-05-04:
The only reason I put the `ifdef` for 4.3 in this [commit](https://github.com/introlab/rtabmap/commit/3e9cff40534c4c9f2c3f58914315d54ed194b239#diff-fb89ec965ee264c1c2edd594c367d6aa) is because I didn't test with previous opencv versions (I assumed that they would compile as they compiled before, I think it is because this file has been recently included in StereoCameraModel.cpp). `core_c.h` is quite standard even from OpenCV 2, so I don't think it will break something else keeping it here inside the ifdef version 4.

commit 7041d5fd (2020-05-04): Fixed #541