introlab/rtabmap #542: vins-fusion and RTABMAP with the support of Realsense D435i

getupcurry, 2020-05-04:
I wonder whether it could be used in turtlebot navigation. As turtlebot  works in plane, so the navigation of turtlebot is based on 2d navigation of ROS.   I wonder  whether the odometry comes from vins-fusion could be used as 2d odometry?   As I know, there is a parameter in RTABMAP  named  "Reg/Force3DoF" , does it could be applied to realize it?

getupcurry, 2020-05-23:
Hello, sir. Do you know any about this? Please help me.

matlabbe, 2020-05-25:
No, VINS-Fusion is a 6DoF approach. `Reg/Force3DoF` would force only graph optimization to be 3DoF, but odometry poses would be 6DoF. `Reg/Force3DoF` is supported only for F2F and F2M odometry approaches. Implementation of VINS inside rtabmap is here: https://github.com/introlab/rtabmap/blob/master/corelib/src/odometry/OdometryVINS.cpp