introlab/rtabmap #543: cannot import point cloud

3DMiller, 2020-05-06:
Is there any way to import a point cloud (.ply) to build a mesh from? Thanks.

3DMiller, 2020-05-12:
> No. From a meshing point of view, you may try meshlab that has also the poisson reconstruction approach used by rtabmap. See for example this video: https://www.youtube.com/watch?v=N3YlncWbMto <https://www.youtube.com/watch?v=N3YlncWbMto>

Thank you. I will ty that. 

I cannot seem to figure out how to upgrade to the lates 0.19 release. Are there instructions for this? I have 0.18 stable release.

Thanks,

Spencer Miller
Masters student in Marine Science - Hawaiʻi Pacific University 
www.millerspencer.weebly.com <http://www.millerspencer.weebly.com/>
www.sketchfab.com/S_Miller <http://www.sketchfab.com/S_Miller> 
https://jmadinlab.github.io/people/spencer <https://jmadinlab.github.io/people/spencer>

matlabbe, 2020-05-13:
Hi, you can get latest windows binaries here: https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts

3DMiller, 2020-05-14:
I am running on MacOS.

matlabbe, 2020-05-26:
You would have to build from [source](https://github.com/introlab/rtabmap/wiki/Installation#source). The instructions may be outdated, looking at the current ports available with MacPort, you may try this:
```
sudo port install vtk +qt5
sudo port install libpcl
sudo port install opencv4
```
You would have all minimum dependencies to build rtabmap afterwards.