introlab/rtabmap #546: Unable to run with Ubuntu Bionic

Zumbalamambo, 2020-05-11:
I tried to run the app through `./rtabmap` but it always says, "Segmentation fault"

why is that so?

Zumbalamambo, 2020-05-11:
Here is the log,

```
> 	linux-vdso.so.1 (0x00007fffd6d59000)
> 	librtabmap_gui.so.0.20 => /usr/local/lib/librtabmap_gui.so.0.20 (0x00007f8cb324e000)
> 	librtabmap_core.so.0.20 => /usr/local/lib/librtabmap_core.so.0.20 (0x00007f8cb2909000)
> 	librtabmap_utilite.so.0.20 => /usr/local/lib/librtabmap_utilite.so.0.20 (0x00007f8cb26e8000)
> 	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f8cb1ea1000)
> 	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f8cb1756000)
> 	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f8cb13cd000)
> 	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f8cb11b5000)
> 	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f8cb0dc4000)
> 	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f8cb0bbf000)
> 	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f8cb09a0000)
> 	libpcl_common.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_common.so.1.8 (0x00007f8cb06e4000)
> 	libpcl_io.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.8 (0x00007f8cb0209000)
> 	libpcl_search.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_search.so.1.8 (0x00007f8cafe02000)
> 	libpcl_surface.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.8 (0x00007f8caf68d000)
> 	libpcl_visualization.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.8 (0x00007f8caf30f000)
> 	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f8caef71000)
> 	libvtkIOMPIImage-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOMPIImage-6.3.so.6.3 (0x00007f8caed49000)
> 	libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 (0x00007f8caeb3f000)
> 	libvtkRenderingLOD-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.3.so.6.3 (0x00007f8cae926000)
> 	libvtkRenderingMatplotlib-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.3.so.6.3 (0x00007f8cae715000)
> 	libvtkRenderingVolumeOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGL-6.3.so.6.3 (0x00007f8cae46e000)
> 	libQt5Svg.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007f8cae219000)
> 	libopencv_imgcodecs.so.4.3 => /usr/local/lib/libopencv_imgcodecs.so.4.3 (0x00007f8cadfb2000)
> 	libopencv_calib3d.so.4.3 => /usr/local/lib/libopencv_calib3d.so.4.3 (0x00007f8cadbf3000)
> 	libopencv_imgproc.so.4.3 => /usr/local/lib/libopencv_imgproc.so.4.3 (0x00007f8cabe7e000)
> 	libopencv_core.so.4.3 => /usr/local/lib/libopencv_core.so.4.3 (0x00007f8caac7a000)
> 	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007f8caaa0b000)
> 	libvtkGUISupportQt-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkGUISupportQt-6.3.so.6.3 (0x00007f8caa7c1000)
> 	libvtkRenderingOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3 (0x00007f8caa321000)
> 	libvtkInteractionStyle-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkInteractionStyle-6.3.so.6.3 (0x00007f8caa09b000)
> 	libvtkRenderingAnnotation-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingAnnotation-6.3.so.6.3 (0x00007f8ca9d51000)
> 	libvtkRenderingFreeType-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeType-6.3.so.6.3 (0x00007f8ca9a8e000)
> 	libvtkRenderingCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingCore-6.3.so.6.3 (0x00007f8ca960e000)
> 	libvtkFiltersSources-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkFiltersSources-6.3.so.6.3 (0x00007f8ca934d000)
> 	libvtkFiltersGeneral-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkFiltersGeneral-6.3.so.6.3 (0x00007f8ca8e1e000)
> 	libvtkCommonExecutionModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonExecutionModel-6.3.so.6.3 (0x00007f8ca8b77000)
> 	libvtkCommonDataModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonDataModel-6.3.so.6.3 (0x00007f8ca861e000)
> 	libvtkCommonMath-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonMath-6.3.so.6.3 (0x00007f8ca83fe000)
> 	libvtkCommonCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.3.so.6.3 (0x00007f8ca7ede000)
> 	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f8ca7775000)
> 	libopencv_stitching.so.4.3 => /usr/local/lib/libopencv_stitching.so.4.3 (0x00007f8ca74c1000)
> 	libopencv_aruco.so.4.3 => /usr/local/lib/libopencv_aruco.so.4.3 (0x00007f8ca7255000)
> 	libopencv_xfeatures2d.so.4.3 => /usr/local/lib/libopencv_xfeatures2d.so.4.3 (0x00007f8ca6d93000)
> 	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007f8ca6b34000)
> 	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f8ca6917000)
> 	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f8ca660e000)
> 	libpcl_kdtree.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so.1.8 (0x00007f8ca62d3000)
> 	libpcl_filters.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_filters.so.1.8 (0x00007f8ca58b9000)
> 	libpcl_features.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_features.so.1.8 (0x00007f8ca340e000)
> 	libpcl_sample_consensus.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.8 (0x00007f8ca2567000)
> 	libpcl_segmentation.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so.1.8 (0x00007f8ca1644000)
> 	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007f8ca13ce000)
> 	libg2o_core.so => /usr/local/lib/libg2o_core.so (0x00007f8ca116b000)
> 	libg2o_types_slam2d.so => /usr/local/lib/libg2o_types_slam2d.so (0x00007f8ca0f06000)
[... cut ...]
> 	libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8 (0x00007f8c8090b000)
> 	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007f8c806d5000)
> 	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007f8c804bf000)
> 	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007f8c80296000)
> 	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007f8c80087000)
> 	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007f8c7fe3d000)
> 
```

Zumbalamambo, 2020-05-11:
Here is the gdb log,

```
> (gdb) r
> Starting program: /usr/local/bin/rtabmap 
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> 
> Program received signal SIGSEGV, Segmentation fault.
> __GI___libc_free (mem=0x20) at malloc.c:3103
> 3103	malloc.c: No such file or directory.
> 
```

matlabbe, 2020-05-12:
You have OpenCV and g2o built from source. Normally OpenCV should be okay, unless you would want to launch rtabmap_ros. For this error with the standalone version, it seems an Eigen problem. This seg fault would be caused by how g2o is compiled. As you are using PCL binaries,  build/install g2o with `-DBUILD_WITH_MARCH_NATIVE=OFF`. Which version of g2o are you using?

closed 2020-08-14 by Zumbalamambo