introlab/rtabmap #547: Installing rtabmap on non-Tango devices (Huawei P30 Pro)

kaatrasa, 2020-05-11:
Hey, I noticed recent commits added support to ARCore & AREngine. I tried to follow https://github.com/introlab/rtabmap/wiki/Installation#docker, but I am new to docker so I had some trouble.

docker run --privileged -i -t --rm -v /dev/bus/usb:/dev/bus/usb introlab3it/rtabmap:android24 adb install -r rtabmap-tango/build/arm64-v8a/app/android/bin/RTABMap-debug.apk

adb: failed to stat rtabmap-tango/build/arm64-v8a/app/android/bin/RTABMap-debug.apk: No such file or directory

Would it be possible to add the apk-file like in introlab3it/rtabmap:tango or can you give instructions how to build the apk myself? Thank you!

matlabbe, 2020-05-13:
Hi,

There was a problem with the docker image. I [fixed](https://github.com/introlab/rtabmap/commit/09d2f7d2326d58f43d640120cd98b38f1271d14a) the build for android. The next image should have the APK built correctly.

There are some problems in dockerhub, images are not automatically built. I'll check to fix that problem too. In the meantime, you can use the image you have, just update rtabmap inside and rebuild:

```bash
$ docker run -it introlab3it/rtabmap:android24
$ cd rtabmap-tango/build/arm64-v8a
$ git pull
$ make
```

cheers,
Mathieu

kaatrasa, 2020-05-13:
Thanks, got it working now!

closed 2020-05-13 by kaatrasa

tangsipeng, 2020-07-02:
@matlabbe matlabbe
i pull docker 
docker pull introlab3it/rtabmap:android24

try to make but fail

>[ 88%] Building CXX object app/android/jni/CMakeFiles/NativeRTABMap.dir/scene.cpp.o
In file included from /root/rtabmap-tango/app/android/jni/jni_interface.cpp:31:
In file included from /root/rtabmap-tango/app/android/jni/RTABMapApp.h:36:
In file included from /root/rtabmap-tango/app/android/jni/scene.h:41:
/root/rtabmap-tango/app/android/jni/background_renderer.h:24:10: fatal error: 'arcore_c_api.h' file not found
#include <arcore_c_api.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
app/android/jni/CMakeFiles/NativeRTABMap.dir/build.make:62: recipe for target 'app/android/jni/CMakeFiles/NativeRTABMap.dir/jni_interface.cpp.o' failed
make[2]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/jni_interface.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /root/rtabmap-tango/app/android/jni/RTABMapApp.cpp:30:
In file included from /root/rtabmap-tango/app/android/jni/RTABMapApp.h:36:
In file included from /root/rtabmap-tango/app/android/jni/scene.h:41:
/root/rtabmap-tango/app/android/jni/background_renderer.h:24:10: fatal error: 'arcore_c_api.h' file not found
#include <arcore_c_api.h>
         ^~~~~~~~~~~~~~~~
In file included from /root/rtabmap-tango/app/android/jni/scene.cpp:32:
In file included from /root/rtabmap-tango/app/android/jni/scene.h:41:
/root/rtabmap-tango/app/android/jni/background_renderer.h:24:10: fatal error: 'arcore_c_api.h' file not found
#include <arcore_c_api.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
app/android/jni/CMakeFiles/NativeRTABMap.dir/build.make:134: recipe for target 'app/android/jni/CMakeFiles/NativeRTABMap.dir/scene.cpp.o' failed
make[2]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/scene.cpp.o] Error 1
/root/rtabmap-tango/app/android/jni/CameraMobile.cpp:203:9: warning: 'setDepthOrRightRaw' is deprecated: Use setRGBDImage() or setStereoImage() with clearNotUpdated=false or removeRawData() instead. To be backward compatible, this function doesn't clear compressed data. [-Wdeprecated-declarations]
                        data.setDepthOrRightRaw(rtabmap::util2d::fastBilateralFiltering(data.depthRaw(), bilateralFilteringSigmaS, bilateralFilteringSigmaR));
                             ^
/root/rtabmap-tango/corelib/include/rtabmap/core/SensorData.h:193:2: note: 'setDepthOrRightRaw' has been explicitly marked deprecated here
        RTABMAP_DEPRECATED(void setDepthOrRightRaw(const cv::Mat & image), "Use setRGBDImage() or setStereoImage() with clearNotUpdated=false or removeRawData() instead. To be backward compatible, this function doesn't clear compressed data.");
        ^
/root/rtabmap-tango/corelib/include/rtabmap/core/RtabmapExp.h:42:60: note: expanded from macro 'RTABMAP_DEPRECATED'
#define RTABMAP_DEPRECATED(func, msg) func __attribute__ ((deprecated(msg)))
                                                           ^
1 error generated.
app/android/jni/CMakeFiles/NativeRTABMap.dir/build.make:110: recipe for target 'app/android/jni/CMakeFiles/NativeRTABMap.dir/RTABMapApp.cpp.o' failed
make[2]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/RTABMapApp.cpp.o] Error 1
1 warning generated.
CMakeFiles/Makefile2:419: recipe for target 'app/android/jni/CMakeFiles/NativeRTABMap.dir/all' failed
make[1]: *** [app/android/jni/CMakeFiles/NativeRTABMap.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
root@e1e283cc5591:~/rtabmap-tango/build/arm64-v8a# vi /root/rtabmap-tango/app/android/jni/background_renderer.h

tangsipeng, 2020-07-02:
i follow rtabmap.bash  and soloved