introlab/rtabmap #549: Is android application database compatible with any other platform?

kaatrasa, 2020-05-13:
Hey, I recorded database using the android application (Huawei P30 Pro) and tried to use it with ros melodic rtabmap. I should have the latest version:
[image]

But when importing database collected by the android application:
[image]

Similarly the android application crashes when importing database recorded using ros melodic rtabmap. Is there some version of rtabmap that is compatible with the android application? Thanks!

matlabbe, 2020-05-13:
The android version is cutting edge version, which is now 0.20. To open it in rtabmap on desktop, rtabmap 0.20 is also required. You have to [build rtabmap from source](https://github.com/introlab/rtabmap_ros#build-from-source), which won't be complicated if you have already installed `ros-melodic rtabmap-ros`.

Bringing a database created from a robot on the phone has not being tested. The rendering in the android app assumes data taken from the phone.

kaatrasa, 2020-05-14:
Thanks again, databases work both ways now: I can import database from ros-melodic-rtabmap-ros to the application and vice versa.

closed 2020-05-14 by kaatrasa