introlab/rtabmap #556: Processing raw image sequence

Zumbalamambo, 2020-05-29:
How do I import a raw image sequence and create a point cloud from the same?

tkircher, 2020-05-29:
What kind of image?

Zumbalamambo, 2020-05-29:
@tkircher the images are raw RGB Image sequence

tkircher, 2020-05-30:
You can use ffmpeg to combine images into a video file.

Zumbalamambo, 2020-05-30:
I have combined the images to a video. Could you please share how I can create a point cloud on the video using orb features?

tkircher, 2020-05-30:
Go to Preferences, select RGB as a source, then in the RGB pane select Video file as a source type.

Zumbalamambo, 2020-05-30:
@tkircher thank you very much!!... I tried it but it throws the following error,

`[ERROR] (2020-05-30 13:18:23) CameraVideo.cpp:128::rtabmap::CameraVideo::init() Parameter "rectifyImages" is set, but no camera model is loaded or valid.`

May I know how to fix this?

tkircher, 2020-05-30:
Try unselecting the rectify images button.

Incidentally, here's a link to a tutorial:

https://github.com/introlab/rtabmap/wiki/Loop-closure-detection

Note that you actually can select images as a source. It looks like they may need to be sequentially numbered in some way.

Zumbalamambo, 2020-05-30:
@tkircher I tried it.. It says, "Poses are null! Cannot export/view clouds" when I try to export the point cloud :(

tkircher, 2020-05-30:
Keep trying! I know it's possible, but I haven't done it with RGB images/video myself.

Zumbalamambo, 2020-05-30:
@tkircher @matlabbe  it just dont give the point cloud after repeated tries. I have also tried to use calibration file and selected rectify images but still the same issue :(

matlabbe, 2020-06-02:
RTAB-Map is not a Monocular SLAM approach. You cannot get a point cloud from only RGB images. For the sequence of image, you don't need to convert to a video, there is an input type for a sequence of images. But as I said, only loop closure detection can be tested with only single RGB frames.

Zumbalamambo, 2020-06-02:
@matlabbe thank you

closed 2020-06-02 by Zumbalamambo

kripper, 2024-11-23:
> RTAB-Map is not a Monocular SLAM approach

Is monocular SLAM still not supported?

matlabbe, 2024-11-24:
Yeah, still not supported. 

There is a way to use monocular camera in rtabmap for loop closure detection, but you will need accurate odometry estimated by other means (like an external visual inertial odometry approach) and enable `Mem/StereoFromMotion=true`. If 3D visual features are produced already by that VIO approach, even better, you can feed these 3D features along the monocular image to rtabmap. This is what we do for android and ios apps without lidar/tof camera, odometry and 3D features are estimated by ArCore or ArKit. You may find some open source VIO approaches that may work on your drone.