introlab/rtabmap #560: Pose jumping?

dschnabel, 2020-06-07:
Found this old issue (https://github.com/introlab/rtabmap/issues/409) and I see that @matlabbe used to have a similar issue like I'm facing now.

I'm also using the Realsense T265 with rtabmap and am seeing pose jumps (see https://github.com/IntelRealSense/librealsense/issues/6532). I did disable relocalization and pose jumping but for some reason the pose still jumps. Would this be an issue with the T265 or rtabmap?

Side note: I wrote my own ROS wrapper for the T265 and the D435 and am sending pose and image data via ROS messages.

dschnabel, 2020-06-09:
I did some further analysis and it appears to be a problem with the transform that rtabmap performs (map -> world).

Here's my transform tree: [frames.pdf](https://github.com/introlab/rtabmap/files/4750366/frames.pdf)

From what I understand rtabmap updates the world coordinates on every detection. I have a detection rate of 1s and you can see nicely from this video that the coordinate update (graph) correlates with the detection log output (console):
[image]

The problem as you see is that while the pose data is stable (see `/tf/t265_pose_frame/translation/x`) the world coordinate jumps around quite a bit (see `/tf/world/translation/x`).

Any ideas what I can try to fix this behavior?

dschnabel, 2020-06-10:
One idea that came to my mind is that I was using a relatively low resolution for the color and depth streams (424x240). Because of that I was theorizing that rtabmap cannot confidently determine the current position.

So I increased the resolution to 640x360 and repeated the experiment. But this didn't seem to make a difference. I've uploaded my map [here](https://dschnabel-uploads.s3-us-west-2.amazonaws.com/rtabmap.db.tar.gz). Maybe something is wrong with the images in the map?

dschnabel, 2020-06-22:
I got better behavior (less jumps) by increasing `Vis/MinInliers` from 20 to 100. Closing this issue.

closed 2020-06-22 by dschnabel

matlabbe, 2020-07-06:
Increasing `Vis/MinInliers` helps to get more accurate loop closure transformations. However, you may get less loop closures to higher it is. There is then a trade-off between accuracy and the number of loop closures detected.

Using /map->/world is difficult to interpret to see the actual jumps. It is better to look at the /map -> /t265_pose_frame tf