introlab/rtabmap #561: Can I run the RTAB-MAP by GPU? I hope it can run faster on TX2 .

weseu, 2020-06-08:
Can I run the RTAB-MAP by GPU? I hope it can run faster on TX2 .
Has any way to let the RTAB-MAP CPU code converse into GPU code.

matlabbe, 2020-06-23:
There is no explicit GPU options for RTAB-Map. You may take a look at OpenCV and PCL libraries for optimization options they can provide. Probably now outdated, but there were some opencv optimizations for Tegra available in jetpack, see https://github.com/introlab/rtabmap_ros/#build-from-source-for-nvidia-jetson

Other approach is to use smaller images, depending on the camera you are using.

weseu, 2020-06-29:
So...It didn't have cuda code now??

tkircher, 2020-07-02:
@weseu There is a significant performance difference in RTAB-Map between the entire support chain (Eigen, OpenCV, VTK, and especially the feature detectors) being built with CUDA support versus without. When you make use of a library that uses CUDA, you are using CUDA.

pramodwic1, 2022-08-24:
Hi @tkircher and @matlabbe ,

I have been working on installing Rtabmap on Xavier NX (with intel realsense D455), following your advice on  (https://github.com/introlab/rtabmap/issues/427#issuecomment-1000921742). As I remember I installed Opencv and VTK with CUDA support. However I built PCL without CUDA.

I managed to get rtabmap working correctly, however when running rtabmap I noticed a significant lag while mapping the 3D point cloud, seems like the lag increases with time to the point where it may take 5 seconds for the 3D point cloud to update to new camera position.

Could I know your experience in this case? It would be helpful if there is any suggestion that can help me run rtabmap in real time. 

Thanks a lot :)

tkircher, 2022-08-24:
@pramodwic1 I recommend running a profiler to figure out where the CPU is spending so much of its time.

tugbakara, 2022-08-31:
Hi, @tkircher I am planning to pass GPU side of rtabmap but in PCL side I didn't find to where to look or how to look of CUDA support for PCL. I am using Nvidia Jetson AGX.

tkircher, 2022-08-31:
@tugbakara There's documentation online for this, see the below link. It essentially requires rebuilding most of the basic libraries from source. When I build rtabmap for Xavier NX, I have to essentially wipe the system and start with NVidia's BLAS and LAPACK, and build Eigen and SuiteSparse and FFTW, all the way up, from source. Otherwise it simply won't work as well as it can. I've benchmarked and profiled rtabmap with different combinations of prebuilt system libraries and there just isn't any other way.

https://pointclouds.org/documentation/tutorials/gpu_install.html

tugbakara, 2022-08-31:
You mean I have to build not only PCL and openCV also the libs that you listed have to be installed with CUDA support and then rtapmap run ,right?

tkircher, 2022-08-31:
> You mean I have to build not only PCL and openCV also the libs that you listed have to be installed with CUDA support and then rtapmap run ,right?

Yes.

tugbakara, 2022-09-01:
@tkircher , thank you so much! :)

tugbakara, 2022-11-22:
> SuiteSparse

Hi @tkircher again! 
You mentioned libs for example BLAS,LAPAS,SuiteSparse etc... Is there any reference for them? Or just you searched in github for their source codes to build or is there anything special , if there is could you explain which steps and order did you follow ? I am a bit newbie for this task and when I made a search about it actually there wasn't anything clear for me.
Thanks :)

matlabbe, 2024-09-04:
Some parts are ported to Cuda right now https://github.com/introlab/rtabmap/pull/1330